jankae/LibreVNA #1: Hardware Schematic/Board conversion to KiCad 5.x

bvernoux, 2020-09-27:
Your project is just amazing and It will even be better if the Eagle files could be converted (and fixed) to KiCad 5.x
I will check that on my side but anyone interested in doing the conversion are welcome

jankae, 2020-09-28:
If you try the conversion be sure to use the newest eagle files (small differences to the pictured PCB). Last time I tried KiCad I didn't like it that much but that was a couple of years ago and from what I heard it improved a lot since then. If someone can pull off the conversion, I'll seriously consider switching from Eagle to KiCad on this project

chrisgj198, 2020-09-28:
Could you please include at least a .pdf of the schematic, for us Kicad users? I think none of my machines have Eagle installed any more.

jankae, 2020-09-28:
Of course, here you go: 294855a

bvernoux, 2020-09-30:
Does some people are interested by the conversion to KiCad 5.x (or even 6.x) or I'm the only one ?

jankae, 2020-09-30:
I'd like the chance to take another look at KiCad, it is just not that high on my priority list. At the moment I am fixing small mistakes/making tiny improvements on the hardware side and I am going to do that with Eagle. I do realize that this board is not covered by the free version of Eagle and for future development switching to KiCad would probably help a lot of people.

I am not going to touch the PCB files for a while after I have made the changes for the next revision (will probably take another week or so) and I think that would be a good time to try a KiCad conversion. But as I don't have any experience with KiCad I am not sure of how much help I can be.

nbgsmk, 2020-09-30:
Personally I never used KiCad, but I see no reason why someone experienced wouldn't be able to try with the existing board, just to see the problems and quirks, and to perfect the process, in the meantime. I wish I could help but I'd probably be spending all time on my own learning curve.

bvernoux, 2020-10-01:
@jankae I'm waiting your feedback when the schematic/board will be frozen I can do the conversion to start on a clean 1st version with KiCad

bvernoux, 2020-10-03:
For information I have filled 2 issues in KiCad nightly related to Eagle 9.3.1 import:
- Eagle 9.3.1 import unsupported Eagle layer 'tRestrict' (41), 'bRestrict' (42) and 'Milling' (46)
   - https://gitlab.com/kicad/code/kicad/-/issues/5862 
- Eagle 9.3.1 import with PCBnew freeze during DRC Control
   - https://gitlab.com/kicad/code/kicad/-/issues/5863

jankae, 2020-10-07:
@bvernoux I finished making the small changes to the PCB and most likely will not touch the Eagle-files for at least a few weeks or even months. Regarding the unsupported layers:
* tTest: Only contains the coordinates of the testpoints, not needed for correct board files
* tRestrict/bRestrict: I used these layers to keep the ground plane away from a couple of RF traces. In Altium I would have used a specific clearance rule between the nets and polygons but as far as I know Eagle does not support that. If KiCad can be configured to do that with DRC rules, there is no need for these layers
* Milling: Only used for pads on the USB-C and power connector (weird work-around because Eagle does not support arbitrary pad shapes)

What is the reason for using KiCad 6.x instead of 5.x? Are there any required features that are only available in the 6.x version? As 6.x is not released yet, I would've expected to use 5.x, but then again I don't really know anything about KiCad, maybe there are good reasons.

I tried importing the project into KiCad a couple days ago and was quite impressed how easy that seemed to be. Apart from the unsupported layers, I think the removed soldermask between the RF sections wasn't imported correctly (only the outline was visible in KiCad), otherwise it looked good at first glance.

sp9bsl, 2020-10-07:
Hi Jan, 
I'm very impressed with your work. I'm considering ordering one small batch of pcbs for private use. You mention about Altium - do you have the pcb/schematics files designed in Altium?

Slawek

jankae, 2020-10-07:
Hi Slawek,
thank you for the compliment, if you do replicate this project please send me pictures/feedback. There are no Altium files, I just used that as an example because thats what I am used to at work. At the moment the only PCB files for this project are created in Eagle.

sp9bsl, 2020-10-07:
Of course, will send pictures. I will wait some time watching what Hugen will do. The ready to buy working version is better option - less time consuming :)

bvernoux, 2020-10-07:
> @bvernoux I finished making the small changes to the PCB and most likely will not touch the Eagle-files for at least a few weeks or even months. Regarding the unsupported layers:
>
>     * tTest: Only [...]

I will check to do it on KiCad 5.1.7-1 (latest stable version), so far I have tried on KiCad 6.x nightly build because there is some nice features I have tested... but the aim is to do it for KiCad 5.1.7-1 as it is very stable.

> I tried importing the project into KiCad a couple days ago and was quite impressed how easy that seemed to be. Apart from the unsupported layers, I think the removed soldermask between the RF [...]

Yes the import is already very nice and with some minor tweak it is fully usable.
I have already the full schematic converted to KiCad 6.x with 0 Warnings and the Board is not far to be very nice (some part shall be fixed but I have noted everything).
I will do that on KiCad 5.1.7-1 with the latest version I have found some very nice hints for the shielding to create some masks from the 3D step parts using FreeCad => InkScape => Paint.net (or any open source tool) => KiCad, I will document it more for others.
Example of actual conversion on Kicad 5.99 nightly (5.99.0-3741-gb38cef409)

Schematic:
[image]
[image]

Board:
[image]

3D Board Preview Top:
[image]

3D Board Preview Bottom:
[image]

bvernoux, 2020-10-21:
If you are interested I'm not far to release the schematic/board files for KiCad 5.1.7-1 especially with hint how to extract the KiCad Mask from STEP file for the shielding using only open source tools of course

kf4mot, 2020-10-21:
Stepup may be a good tool for this project if someone is interested in some Kicad/Freecad integration. https://kicad-pcb.org/external-tools/stepup/

jankae, 2020-10-22:
@bvernoux I am definitely interested in the KiCad files, thank you doing the conversion. Unless I find KiCad extremely awkward to use after spending a bit of time with it, I will eventually move away from Eagle on this project.

bvernoux, 2020-10-25:
Conversion done see https://github.com/jankae/VNA2/pull/7

jankae, 2021-04-17:
As KiCad does not support round soldermask regions for now, I decided to stay with Eagle. I'll check the KiCad features from time to time (as some of the required stuff is being worked on, see e.g. https://gitlab.com/kicad/code/kicad/-/merge_requests/731) and once it supports everything I will gladly switch to it

closed 2021-04-17 by jankae