jankae/LibreVNA #3: VNA build

nbgsmk, 2020-09-28:
Hello again!
I would like to express, again, my excitement with your project! Thanks for this professional instrument.
Regarding the physical build, let me start with several questions. Please have a bit of patiece :-) as there are some parts of the process which I am not familiar with. Tne, kindly correct or add info where needed:
- you have ordered the board at Jlcpcb?
- thickness: 1.6mm
- impedance control: yes, jlc7628
- surface finish, judging by the photo: ENIG
- copper weight: 1oz

Then, a few more: 
- if I provide .step files to the metal shop, this is enough to completely produce the shield?
- finally, and most important, since the board exceeds the free licence limitation for Eagle: if the board design is fixed now, could you _please_ provide the gerber files? Thanks!

Just out of curiosity, did you use Jlcpcb's assembly process or did you populate the board yourself?

jankae, 2020-09-28:
Hi again, I hope these answers help:
The manufacturing process was a little bit different for the first version (github.com/jankae/VNA) and the current prototype.
First version:
* I ordered the PCB at Jlcpcb, your settings are correct, except that I didn't use ENIG (but would recommend it for you)
* Everything was populated by hand
* I milled the enclosure myself but the step files are all that is needed (along with the instruction to tap the drilled holes)

Current prototype (this repository):
* While the VNA is designed by me, the manufacturing was handled by Hugen (the one associated with the nanoVNA project). The PCBs are once again ordered from Jlcpcb but I don't know if he used the assembly process there.
* I can get you the gerber files now but keep in mind that this is still a work on progress. I would recommend you wait at least a couple of weeks before ordering anything, there might be small PCB changes soon.

jankae, 2020-10-14:
You didn't miss anything, I haven't uploaded the gerber files. I can get them to you next week, I don't have access to my computer at the moment.

bvernoux, 2020-10-14:
Could you provide characteristics/PCB used/planned to build the VNA2 ?
- I see on VNA2 design there is some long RF traces and ENIG is not very good for RF (especially the insertion loss/skin effect especially for GCPW and for frequency >2GHz) so Immersion Silver could be better (the best will be ISIG but it is expensive and hard to find a fab for that) on the other side I'm not sure it is good to use Immersion Silver for long term and depending on environments (oxidation/corrosion, formation of silver whiskers...)

sp9bsl, 2020-10-15:
Benjamin, the ENIG in JLC is AFAIK applied to exposed copper after applying solder mask so, only affects the soldering pads. I have other question: in Jan's pcb I see long high frequency traces which are covered by solder mask, isn't it worse thing? I was taught not to use solder mask on CPW traces...

jankae, 2020-10-15:
I have used the JLC7628 stackup in all PCBs. Before building the complete VNA, I made smaller breakout parts with individual sections and connected them with coax. I also experimented with HASL/ENIG finish and tested the removal of solder mask on those PCBs. None of that seemed to make a significant difference, in all cases the amount of solder or exact placement of components had more influence.
In professional RF designs the solder mask is almost always removed but I thought that is only because the dielectric and thickness can't be controlled very well (which is especially important if you implement filters with PCB traces). On lossy FR4 (accurate and sharp) filters are not possible at all and according to my measurements, the solder mask doesn't really contribute anything to the losses.
Unfortunately, I didn't save any of these early measurements and haven't taken any on the VNA prototypes yet.

sp9bsl, 2020-10-15:
Hi Jan,
thank you for this explanation. Please don't think I'm complaining :) I'm very grateful you share the design. As many already wrote - hats off!

hugen79, 2020-10-16:
This is the file I compiled, including GerberFiles, embedded software and linux App.
https://drive.google.com/drive/folders/1cO0HWymSRwd41WROYJd9gooWAj6qmcvp?usp=sharing
I still have a small number of test pcbs made for the first time, but the remaining PCB isolation is poor and can only be used for functional testing. If it is useful to you, you can send a private email to me and I will send it to you.
hugen(at)outlook.com

nbgsmk, 2020-10-18:
@hugen79 Hi! By poor PCB isolation, do you mean the pcb quality is not good regarding rf signal isolation? Or something else?
In both cases, I am just curious how did you determined that? By some measurements in advance, or by poor performance of the completed analyzer?

nbgsmk, 2020-10-22:
Hi @hugen79 !  Somehow I couldn't find your email anywhere, and github has removed the option to directly contact users. Any suggestion? 
Second topic, what is the state of the "poor-isolation" board? Is it populated with all components and otherwise operational? In that case, I'm thinking It would be usefull, we could talk. If not, then I would rather take my chances with a new board from jlcpcb and spend time ordering and assembling that. 
Thanks for your time, in any case :-)

sp9bsl, 2020-10-22:
Hi,
Hugen posted his email four post above ;)

nbgsmk, 2020-10-26:
Hello @jankae !
Seing how conversion to KiCad is still a work in progress, I would appreciate if you could prepare gerber and pick'n-place files from your own process, so to speak.
Although I am in contact with hugen, I would like to try the JLCPCB ordering myself, and use their SMT process for 2 completed boards.
Thanks!
Zoran

jankae, 2020-10-26:
Sorry, I assumed you already went ahead with the Gerber files Hugen provided. I have generated the files from the latest Eagle PCB layout (although they should be identical to Hugens Gerbers):
[VNA_2020-10-26.zip](https://github.com/jankae/VNA2/files/5441123/VNA_2020-10-26.zip)
So far I always assembled my boards by hand, and had to look up how to generate pick-and-place files from Eagle. I hope these work for you:
[PickAndPlace.zip](https://github.com/jankae/VNA2/files/5441130/PickAndPlace.zip)

nbgsmk, 2020-10-27:
Thanks! I will give them a look first thing in the morning!

sulautettu, 2021-03-14:
Hey, big thanks for all your hard work. 
Does these settings seem right? I have only used OSHPARK before and these settings options are quite overwhelming.

[image]

I don't know is these "issue" section for this, but would there be good section to organize group orders?
If some one is ordering PCB's and willling to after distribute to Finland, i would be very interested - If not I'll buy 15 copies and can Re-distripute them if someone needs, at True cost (inital cost/pcb's + shipment)

I have allso homemade CNC-mill and trying to make the case  - I was thining allso to mill small haetsinks at top of the case and place for 40mm pc fan in the middle. If it goes well i can allso supply those.

BR. Karri

jankae, 2021-03-15:
Hi Karri,
yes, they look correct. The most important thing is the JLC7628 impedance stackup you selected.

Pretty sure the github issues are not intended for organizing group orders but feel free to create a new issue for that anyway :) At least I don't know a better place at the moment to do so, this project probably isn't big enough to warrant its own group/forum for such a thing.

FYI: Hugen is currently assembling a first batch for sale but it is taking a bit longer than expected because some parts are hard to source at the moment. But building it yourself is of course more fun and your proposed addition of a fan seems quite useful.

nbgsmk, 2021-03-21:
Hello @jankae !
Just a short question, could you tell if there were any changes to the aluminum case for the VNA? I own a sample of the case, dating back to approx. november. I know about the errata and small pcb corrections in the meantime, but that does not necesarrily change the alu milling part. In other words, being away for a while, I am not sure anymore if my "old" alu-case will fit the latest pcb, if I decided to make one?
Thanks!

jankae, 2021-03-22:
Hi, there has only been one change to the step file that is relevant for the PCB. I added holes for additional screws and also made the PCB and the case 2mm larger. The easiest way is probably to check [this](https://github.com/jankae/VNA2/blob/master/Documentation/DeveloperInfo/VersionsAndModifications.md). If your case has the two screw holes at the front between the ports like the "Second Prototype" PCB, it will still fit on the PCB if you make one with the current data.

nbgsmk, 2021-03-24:
Yes, it fits the second prototype. That's great. Thanks!
I'm not sure if there are other threads ongoing here, so I will not close the issue.
Zoran

closed 2021-04-17 by jankae