jankae/LibreVNA #4: TRL calibration

tchristle, 2020-10-01:
Can TRL calibration be implemented as an alternative to SOLT?  The standards for TRL are easier to define and fabricate.

jankae, 2020-10-01:
I had a look at the TRL procedure and I think that should be possible. As I have never used a TRL calibration before: does [this](http://icmicrowave.com/pdf/ENA_Multiport_TRL_LRM_manual.pdf) describe the typical workflow?

tchristle, 2020-10-01:
Yes, that's a good description of the work flow.  It might be best to shelf this idea for way later because TRL is not practical below 1GHz.  The reason being is that the line standards need to be greater than a quarter wavelength.  So if someone wanted a mictrostrip kit that went down to 200MHz then they would need an 8 inch microstrip line and who wants to build that.  I assume most interest  in this project will be using it at lower frequencies.

jankae, 2020-10-03:
From what I understood, the phase shift of the line can be anything from 20-160 degrees for acceptable results. I just added TRL calibration and tried it with a 5cm line and a sweep from 200MHz to 1GHz. Checking the calibration by measuring some attenuators showed good results.

So far, it is limited to only one line standard, so it only works over a limited frequency range. Adding a second line standard is probably not much work but what I really would like to do is combine it with TRM calibration for lower frequencies (as described in the document I linked to).
However, I couldn't find any document describing the formulas for that and a short attempt to derive them myself failed. Does anyone know where I can find the formulas for the error correction terms for a TRM calibration?

tchristle, 2020-10-04:
If you're only looking at magnitude response to any of your verification standards then you'll always find reasonable agreement with your expectation.   You'll identify the calibration falling apart of you scrutinize the phase response more closely.   Especially on the reflection coefficient.   I think Beatty lines are the easiest standard to use for verification.   Keysight has a good write up on this.   https://www.google.com/url?sa=t&source=web&rct=j&url=http://www.keysight.com/upload/cmc_upload/All/How_Good_is_Your_SI_Simulator.pdf&ved=2ahUKEwizkOuzwJvsAhVYOs0KHa7SB8sQFjADegQIDBAB&usg=AOvVaw0JMjUtpe5mWEGI_4pC32F2

jankae, 2020-11-08:
The TRL calibration has been confirmed working by someone using an APC7 airline, I guess it is okay to close this now

closed 2020-11-08 by jankae