jankae/LibreVNA #6: Where to from here

hedleyd, 2020-10-14:
1. Are the step files for housing available 
2. If so, will the machined parts work on VNA and VNA2 or are they different
3. Where can I download the application program that runs on the PC.
4. I understand this is a beta version in development , so when do you expect a stable release eg 6 months . 
5. I have a small VNWE 3E and normally am using the sub GHZ range for RFID typically center freq of 13.56 MHZ - will your unit work at such a low frequency ? .
6. If anyone is familiar with the VNWA3 , what beside higher frequency would be the advantage of this solution 

Thanks in advance

jankae, 2020-10-14:
1. Yes, the step files can be found in the hardware folder
2. They are completely different. There has also been a change in the step files within VNA2. The newest step files don't match the currently pictured PCB exactly (but they do match the current PCB layout)
3. Either from the [releases](https://github.com/jankae/VNA2/releases) or from the actions-tab if you want the newest version built from the latest commit
4. Probably a couple of months, but I don't really know for sure (depends on how much time I can make for this project). The current version is working reasonable well
5. It does work at 13.56 MHz. There is no fixed lower limit, you can even go below 1MHz but the performance won't be as good (lower stimulus level, less dynamic range)
6. Sorry, I don't know that VNA. My guess would be that there are not many (if any) advantages when only working with low frequencies. Maybe a bit more dynamic range but that depends on the final specifications of my device

hedleyd, 2020-10-14:
Many thanks for prompt reply
You can download the manual and PC software for VNWA from -
https://www.sdr-kits.net/DG8SAQ-VNWA-software-documentation-user-guide
This may be of intrests when extending you functionality
Ill check in again next year and then download and start building your VNA2
- PCB looks really well laid out from an RF perspective and case again
looks fantastic.
I will download the step files and , if I understand correctly should wait
for a stable PCB before machining . I will then align PCB and step model in
Cadence Allegro before generating CNC gcode files.

All the best with this exciting project

Hedley Davidson
Managing Director
Cynaps Technologies Pty Ltd
hedley@cynaps.co.za
078-458-7476 ( Mobile )

jankae, 2020-10-19:
Thanks for the link to the manual, I'll read through it and check for useful functionalities.

If you are not in a hurry, it is probably best to wait a bit before replicating this project. I don't expect significant changes to the hardware but small improvements might still happen.

chrisgj198, 2020-10-26:
I think this Jankae design is much better than the VNWA (and VNWA3 if it is similar), though I have not seen a full schematic of the VNWA3 version. The SA612 mixers (aside from frequency limitations) don't have degeneration on the RF input stage so would have poor linearity and gain stability vs temperature. Also the VNWA used an aliased image coming out of the DDS as the LO signal for higher frequencies, and whilst the desired image frequency is present in the LO from the DAC, also a much stronger fundamental signal is present, and critically, these are not usually harmonically related. They feed the LO port of a mixer and the LO port normally does (and should) have a non-linear transfer function (pretty much hard-switching mixer) and I think the gain experienced by the DAC image may not be well-defined (nor very strong) whilst it is being hard-switched by the fundamental frequency too, leading to unstable measurements. I think they should have used odd-harmonic mixing instead, since the strong fundamental driving the non-linear LO port of the mixer would then be no problem for converting using e.g. the 3rd harmonic of the LO.

hedleyd, 2020-10-26:
Thanks

If you watch from 2:45 - although it is in german the video is easily
followed.
https://www.youtube.com/watch?v=SsbGAH9zdVM

Regards

Hedley Davidson
Managing Director
Cynaps Technologies Pty Ltd
hedley@cynaps.co.za
078-458-7476 ( Mobile )

alextsure, 2020-10-28:
Hello Jankae,
thank You for sharing this great project!
I have the following issue while trying to open FPGA project files in Xilinx ISE.
When the main project file VNA.xise opens, the ISE complains that there are several absent files .xco & .xise  from the absent directory ipcore_dir  in the VNA project folder.
Would You, please, be so kind to upload or share the absent IP core files from the absent dir?
I'm just studying the Xilinx ISE workflow.
Thank You!

jankae, 2020-10-28:
Hm, I guess my gitignore was a bit too aggressive. The files are added now, I was able to open a cloned version of this repository with ISE. If you find something unusual or weird in the VHDL please let me know, I am a bit more comfortable with microcontrollers and not that experienced with FPGAs.

alextsure, 2020-10-29:
Hello Jan, thank You for ISE update.
While the project opens in ISE from now on, but when I try to implement top VHDL module, the following error occurs:
FPGA\VNA\window.vhd" Line 45: Cannot open file 'Hann.dat'
I tried to run the WindowCoefficientGenerator.py script in Spider (v. 3.7) to generate the 3 absent  *.dat files, but unfortunately it only creates 3 empty files (of 0 bytes size). I am not very much fluent with Python, don't know what should be done to get a proper output files

jankae, 2020-10-29:
Try running the script with Python3 (I got empty files with Python2 as well). If that does not work, use this:
[WindowData.zip](https://github.com/jankae/VNA2/files/5458722/WindowData.zip)

hedleyd, 2020-10-29:
I imported the PCB into Cadence Allegro - Perhaps translation issues but it
shows

Layers different sizes - what size in mm is the actual PCB
A few tracks missing - yellow point to point lines
DRC errors with track spacing - I think this is a translation resolution
issue.

I will test against the schematic and then take the plunge and get 4 *
PCB's made for build and test .

Hedley Davidson
Managing Director
Cynaps Technologies Pty Ltd
hedley@cynaps.co.za
078-458-7476 ( Mobile )

alextsure, 2020-10-29:
This issue with Python is solved now. 
When added a w.file.close() 
It's OK now, the Spider saves .dat files to disk

jankae, 2020-10-29:
@hedleyd First, a word of caution: The PCB design on the master branch is not verified yet (meaning I don't physically have it yet). Compared to my current prototype the are only small changes and fixes, so I am relatively confident that it will work, but of course there still is a slight risk. The changes compared to the prototype are:
* Fixed reversed SCK/MOSI lines for FPGA configuration
* Moved track that crossed through RF section border at the bottom (would short to the aluminium ground)
* Added DC jack because not all USB ports are powerful enough to supply the device
* Added additional screws for better isolation
* Added RC lowpass on digital control lines (also hoping for better isolation)

I have never used Cadence Allegro, so I am not sure whether your issues are a problem. The PCB size is 98x122mm. If you only want to build it (and not change anything), I'd recommend to use the Gerber files I've added [here](https://github.com/jankae/VNA2/issues/3#issuecomment-716786031). I believe that is less risky than importing it in other tools and fixing issues.

@alextsure I have modified the python script accordingly and also added the generated window data files to the repository.

hedleyd, 2020-10-29:
Many thanks - gerber will  definitely be the best solution . Just
downloaded and installed Xilinx 10gb tool chain . Steep learning curve
ahead 👍

closed 2021-04-17 by jankae