jankae/LibreVNA #9: Software build / toolchain setup

nbgsmk, 2020-11-19:
Hello Jan!
I am having a bit of struggle to complete the build of PC Application. Could you help, please?
I have these installed
- Qt Creator community version from offline installer
  my help-about is:
    Qt Creator 4.13.2
    Based on Qt 5.15.1 (MSVC 2019, 64 bit)
	
- Libusb (from libusb-1.0.23.7z archive)
  Unpacked to c:\libusb and included in Application.pro like this
    win32:INCLUDEPATH += C:\libusb\include
    win32:LIBS += -LC:\libusb\MinGW64 -lusb-1.0
  This may not be the best way but, at least, it has eliminated build errors in Qt Creator for now.

- Qwt
  Here is the real problem, I think. 
  Your Application.pro indicates qwt-6.1.4
    I followed a few simple installation commands from QWT website:
	qmake qwt.pro
    mingw32-make
    mingw32-make install
    I could not get 6.1.4 to compile with mingw 8.1.0 or mingw 7.3.0 (both came with Qt installation. later I removed 7.3.0 to avoid possible conflicts)
	Compilation breaks with errors in so many places I could not count... :-)
  I switched to qwt-6.1.5
  ...
  This compiles with both mingw versions, but breaks further into the "mingw32-make install"

////
mingw32-make[1]: Entering directory 'C:/qwt-6.1.5/doc'
C:\Qt\5.15.1\mingw81_64\bin\qmake.exe -install qinstall C:\qwt-6.1.5\doc\html C:\Qwt-6.1.5\doc\html
Error copying C:/qwt-6.1.5/doc/html/analogclock.png to C:\Qwt-6.1.5\doc\html\analogclock.png: Cannot open C:/qwt-6.1.5/doc/html/analogclock.png for input
mingw32-make[1]: *** [Makefile:416: install_doc] Error 3
mingw32-make[1]: Leaving directory 'C:/qwt-6.1.5/doc'
mingw32-make: *** [Makefile:163: sub-doc-install_subtargets-ordered] Error 2
////

  Since these seem related to doc folders, I tried to disregard and proceed:
  Added C:\qwt-6.1.5\lib to PATH
  Added env variable QT_PLUGIN_PATH = C:\qwt-6.1.5\lib
  reboot
  Cloned a fresh copy of VNA2 from github.
  Qt Creator -> navigate and open Application.pro
  But build fails at tracesmithchart.h / .cpp with many errors such as
     field 'thinArcsPath' has incomplete type 'QPainterPath'
     field 'thickArcsPath' has incomplete type 'QPainterPath'
  followed by others, probably as a consequence of that.

I tried some changes in Application.pro but no improvement.
Tried searching the web, but no good results. 
Finally I am stuck here. 
Apart from the code from github, and my humble attempts, what else is needed externally?
Could you explain more about your tool versions, setup, paths, env variables etc? 
Attached my compiler output, if it helps...

Thanks!
Zoran

 
[compiler_output.txt](https://github.com/jankae/VNA2/files/5564301/compiler_output.txt)

jankae, 2020-11-19:
Hi Zoran,

your solutions for libusb and Qwt look fine to me (I simply put the libusb files into the mingw include folder, thats why its path isn't added in the Application.pro). For reference, here are my current tool versions on Windows:
* Qt Creator: 4.12.4
* Qt: 5.9.9
* Qwt: 6.1.4
* libusb: 1.0.23
* mingw: 5.3.0

Using github-actions the code also gets compiled against Qt 5.12.9 using mingw 7.3.0.

Looking at your compilation error, it looks loke Qt changed the internal include structure in 5.15. Could you please add
```c
#include <QPainterPath>
```
in tracesmithchart.h and try again? This should fix your current error but of course there could be others. I'll install Qt 5.15 and try it myself later.

nbgsmk, 2020-11-21:
Hi Jan!
You were right, thanks! I added Qt 5.12.9 and mingw 7.3.0 64bit to my installation, as you suggested this works on github. This time both qwt-6.1.4 and 6.1.5 are compiling just fine.

To summarize my setup, it may help others:

- help->about: "Qt Creator 4.13.3 Based on Qt 5.15.2"
- BUT! Qt 5.12.9 - selected as default kit for this app (compiles with only 1 warning)
- mingw: 7.3.0 64bit
- Qwt: 6.1.4 (in c:\qwt-6.1.4 + added env variable QT_PLUGIN_PATH=C:\qwt-6.1.4\lib)
- libusb: 1.0.23 (in c:\libusb-1.0.23)
- Application.pro -> added lines
  win32:INCLUDEPATH += C:\libusb-1.0.23\include
  win32:LIBS += -LC:\libusb-1.0.23\MinGW64\dll -lusb-1.0

PS. Out of curiosity, after reinstalling, again I tried compiling qwt-6.1.4 with mingw 8.1.0 64 bit and it's still failing with errors and warnings about various things. And, now that I know what I am looking for, I also see mention of QPainterPath! Giving up on that, until further notice.

Now, let's go further. I have several questions and suggestions. I will try to implement at least some of them myself, but please, please forgive my lack of experience with Qt. :( Would you like me to open issues for some of these, or contact you via email, or something else? Or just let you work on your own tempo? :-)

Zoran

jankae, 2020-11-21:
Hi Zoran,
glad you got it working. Don't worry about being new to Qt, I'm sure you'll get to know it fast (GUI programming in general was new to me prior to this project as well).
For bugs or improvement suggestions, I prefer opening issues (easier to track what has been fixed than with mails). I believe for most questions, issues might also be the best way to go (that way other people can find the answers as well). If it gets way too specific, we can always move it to mails later.

jankae, 2020-11-23:
Looks like you found your way around the project, I'll close this.
Just a heads-up: I am currently reworking the graphs (and incidentally removing the Qwt dependency) so if you want to work on something maybe stay away from the graphs for now (or use the graph_update branch which will soon be merged into master)

closed 2020-11-23 by jankae

nbgsmk, 2020-11-23:
Okay Jan! 
One interesting thing, I found that qt 5.12.9 does not work for me straight out of the box. For some reason, when initially opening this project after git clone (or after project-clean, build), the App crashes during start (sorry, didn't keep the logfile).

Upon (unclear sequence of) changing between 5.12.9, 5.15.1, 5.15.2, at one moment I get compilation with 5.15.x with over 30 warnings, and then compilation on 5.12.9 with just 3-4 warnings. (Interesting that one of the 32-bit compilers fails completely, so I use only 64-bit versions). After that, I choose to (ignore that weird behaviour and) just run on 5.12.9, as it is the most similar to your/github setup. Seems like some compilation artifacts are not fully cleaned, and they remain when going back to the Qt 5.12.9 version.
I believe this is a consequence of my working on windows, as opposed to Linux, where many things are already existing. Anyway, I don't see any unreasonable behaviour in the app, so I decide to ignore this weirdness.

PS. I don't expect you to troubleshoot this, I am only documenting the issue, as it may help someone else. 
Sorry for commenting after close. I'll try to close it myself now.

jankae, 2020-11-24:
> the App crashes during start (sorry, didn't keep the logfile)

Could it have been this message: QWidget: Cannot create a QWidget without QApplication? I don't remember what the default in QtCreator is, but if you have set the build configuration to "debug" you'll also need to use .dlls that have been build for debug (can't mix release/debug on Windows), otherwise you will get this error on startup and the app crashes.

nbgsmk, 2020-11-26:
Yes, that was the message. Thanks for clarifying! Now, that you mentioned, I remember reading something about not-mixing of release/debug on windows. Looking at the logwindow of Qt Creator, I didn't notice anything that would point me in that direction, or I didn't know how to filter for it, or something...
But honestly I didn't pay much attention to it afterwards, as it now happens to work practically every time. In the beginning, I must have been playing with project setup a lot, and with versions 5.12.x / 5.15.1 / 5.15.2, 32/64bit etc (but with not enough knowledge of the IDE itself :) .  Out of curiosuty, I will try it in the evening. But never mind, forget it... and thanks for the explanation, again.

> the App crashes during start (sorry, didn't keep the logfile)

> > > Could it have been this message: QWidget: Cannot create a QWidget without QApplication? I