jankae/LibreVNA #11: Plot corruption - ratsnest

nbgsmk, 2020-11-25:
Hello Jan!
I noticed a strange bug, apparently device in initialization, when switching modes. I was struggling for two-three days to reproduce it, and finally, here is how:
Start app (goes into vna mode)
(possibly switch to spectrum analyzer but sometimes even without this step)
switch to signal generator
enable port 1
disable port 1
switch to vna mode

After this:
Graphs become like a ratsnest (attached pics)
Only usb cable (re)connection helps, ie the device needs to be rebooted.
Funny thing, I just opened the window for manual control, closed it without making any changes, and things are back to normal.

I am not really sure about the followng, but _maybe_ device temperature plays a role. Currently it is 50/59/34, but in the screenshot I took, it was 28/38/21. I want to say, I am swithching modes very very often, and 'torturing' the vna :-) with all kinds of possible and impossible settings. Just this morning, after using signal gen. mode, and back to vna, nothing unusual happened. Several times. And now, after 2-3 hours, I can reproduce it reliably. I noticed this a couple days ago, but back then, I couldn't reproduce it at all. So I am not sure if temperature has anything to do with it.

I would not consider it a high priority issue, as it happens relatively rarely, and there is an easy way out of it.
I'd try to chase it myself, but my understanding of hardware specifics, registers, and especially fpga, is not enough for independent troubleshooting. Not yet. However, if you take the time for this issue, and have any suggestions and requests for me, I am willing to do anything. :-)

Has anyone else noticed this on their device? I hope other people are reading, too. How are your temperatures under prolonged operation? 

Regardless, here is a suggestion:
- add option in the menu to reboot device, what do you think? Or in the manual control window?

Cheers!
Zoran

[image]
[image]

jankae, 2020-11-25:
This is definitely a firmware bug, it looks like the source does not get enabled when switching modes in the way you described. Without the source, all receivers essentially see no signal (only noise), so all S parameters are really noisy and at 0db on average. Which version (git commit) of the firmware and PC application did you use?

> Just by full init between modes, or a reboot (I'm sure it's lightning fast) and we would never notice it.

There is already a full FPGA reset between modes (noticeable by the "ready" LED turning off shortly). But looking at it now, I think I broke something in one of the latest updates. It is not doing the reset when switching to/from the signal generator. I'll look into it.

jankae, 2020-11-25:
978842a should fix this, let me know if you still see the problem.

> How are your temperatures under prolonged operation?

Up to 60°C after several hours. Already uncomfortable to touch but okay for the electronics. According to report from another user, adding a fan keeps the temperatures below 30°C but personally I prefer the silent fanless operation.

I wasn't aware that you already have the hardware, did you got one of the samples from Hugen? Or did you build it yourself?

OneOfEleven, 2020-11-25:
That rats nest I've seen with my software too. Yes it's a firmware problem. I worked out how to get around it so as not to see it again but that's not a fix for the problem, a firmware fix is the only answer.

nbgsmk, 2020-11-25:
As far as I can find my way around Git in Qt Creator, I checked the latest graph-update branch. 
In paralell, I use IntelliJ IDEA (a beautiful Java IDE from JetBrains) to verify what I am doing. It turns out, you can open the same projects and folders with both of them at the same time and it works flawlessly. What I do in Qt Creator, immediately is shown in IntelliJ and I can confirm all the edits, branches and differences in a very friendly and pretty gui-like interface. To revert to the short answer, I pulled the latest graph-update from this morning. 

Regarding the hardware, I got a sample from kind Mr. Hugen, thanks to him **a thousand times!** 🥇 👍 
I was about to build one or two for myself, but Hugen kindly offered his help. By the interest this has produced in me (decided to install and learn Qt basics just because of this), and judging by the amount of time I spend every day playing with it, and contributing here-and-there, I hope I am justifying his generous intervention, so to speak. :) 
Zoran

nbgsmk, 2020-11-25:
Just tested the latest FW:
- updated to the latest ver., tested the issue as described - problem no longer present
- downgrade to previous - able to reproduce the problem
- upgrade to latest again - issue fixed.
I propose to close this.

closed 2020-11-25 by jankae

nbgsmk, 2020-11-30:
Sorry, could not resist commenting, although closed. 
I ran my VNA for 12 hours straight (!), actively working on it/with it. It was laying on a textile surface, so not very heat conductive.
It got upto maybe ~65°C but not higher. 

> Up to 60°C after several hours.

sp9bsl, 2020-12-01:
Hi,
my device laying on the desk, after about an hour reports around 50°C. Placing a 90mm 12V fan on it, with 5V power supply (almost noiseless) causes temperature to fall around 25..30°C. 

I also have the sample from Hugen (arrived last Friday), I want to thank him for his kindness!

Slawek