jankae/LibreVNA #14: BOM changelist

blackberryer, 2020-11-30:
First of all, it's really, really great，and thanks for your design and sharing.
After looking at the design, I raised the question of component replacement in order to further reduce costs or improve performance（according to each person's own situation）：
1>RFSA3714（50MHz to 6000MHz） to RFSA3713（5MHz to 6000MHz）,Package is not pin to pin
    or to PE43711（9kHz to 6000MHz） ,Package is pin to pin;
2>QPC6324SR（5MHz to 6000MHz） to PE42420（100MHz to 6000MHz） ,Package is pin to pin;
3>B012-617DB-1-63-B331（10MHz to 6000MHz？） to TCM1-63AX+（10MHz to 6000MHz），B012-617DB-1-63-B331 is EASTEVER BALUN，but I can't find the datasheet，can you provide the datasheet？I would be very grateful！
or
MABA-011082（5MHz to 8000MHz）

jankae, 2020-11-30:
Hi, thanks for the suggestions. If you compare the schematic with the earlier version (https://github.com/jankae/VNA) you'll notice that I pretty much changed parts in exactly the positions you mentioned before, when redoing the design for this version.

Some initial thought from my side:
> RFSA3714  to RFSA3713

Although the RFSA3714 is only specified down to 50MHz, it actually works great on my prototypes all the way to 100kHz (going down even further doesn't really make sense because of the limitations of other parts). I guess RFSA3713 or PE43711 would work just as well.

> QPC6324SR to PE42420

If the PE42420 also works at lower frequencies, it could be an option but I don't see the need to change it at the moment. From what I've found,
 the cost is pretty much identical.

> B012-617DB-1-63-B331（10MHz to 6000MHz？） to TCM1-63AX+

The [B012-617DB-1-63-B331](https://github.com/jankae/VNA2/files/5618474/B012-617DB-1-63-B331.pdf) is actually a replacement for the TCM1-63AX+, so yes, that would definitely work as well. The MABA-011082 is what I used in the first prototype and while it is OK, I measured a slightly worse directivity than with the B012-617DB-1-63-B331. The problem with all of these is that you need to remove the tertiary winding, otherwise they won't work very well at all. For prototypes, this can easily be done with tweezers and a steady hand.

nbgsmk, 2020-11-30:
Hi friends!
Far from me being competent to talk about this, at least not without detailed look into the datasheets. Even if I did, it comes down to testing on a real prototype, and testing a lot! Imho, many parts look alike on paper but they begin to differ in real life. While it is cool to say "hey I built myself a VNA which goes down to 9kHz" I'm just considering how often would I use such a thing? Please, correct me if I'm wrong, and I seriously mean it, please do correct me. Maybe I am missing a whole universe of vna applications, and not utilizing it to the full potential.
Being as it is, right now I have an awesome instrument, fulfilling 200% of my needs now and in the future. My only concern today is heat and, in that regard, the only reason I would think of redesigning the vna is to get it a little bit cooler. As you may have noticed from another post, I had it running for 12 hours straight, both for measurements and for finding little software changes to suggest to Jan. And it performed wonderfully! But I am thinking 60°C definitely poses some stress on the pcb, traces and solderjoints. After a year, two, three, of streching and cooling how would it be? Instead of waiting to see, I am definitely going to fit it with a heatsink (no fan). In the meantime, are these suggestions bringing reduction in power consumption? 
Zoran

blackberryer, 2020-12-01:
> Hi friends!
> Far from me being competent to talk about this, at least not without detailed look into the datasheets. Even if I did, it comes down to testing on a real prototype, and testing a lot! Imho, many parts look alike on paper but they begin to differ in real life. While it is cool to say "hey I built myself a VNA which goes down to 9kHz" I'm just considering how often would I use such a thing? Please, correct me if I'm wrong, and I seriously mean it, please do correct me. Maybe I am missing a whole universe of vna applications, and not utilizing it to the full potential.
> Being as it is, right now I have an awesome instrument, fulfilling 200% of my needs now and in the future. My only concern today is heat and, in that regard, the only reason I would think of redesigning the vna is to get it a little bit cooler. As you may have noticed from another post, I had it running for 12 hours straight, both for measurements and for finding little software changes to suggest to Jan. And it performed wonderfully! But I am thinking 60°C definitely poses some stress on the pcb, traces and solderjoints. After a year, two, three, of streching and cooling how would it be? Instead of waiting to see, I am definitely going to fit it with a heatsink (no fan). In the meantime, I am really curious, are these suggestions bringing reduction in power consumption?
> Zoran

Hi nbgsmk
I notice that "My only concern today is heat",which is absolutely correct, and I'll take a warning.

pentti12, 2020-12-01:
Hello

No more overheat issues:  I put  the giant heat sink directly over the aluminum enclosure.  Now the PLL temperatures are 36/36 at room temperature of 25 degrees. Pentti
[image]

nbgsmk, 2020-12-01:
> Hi nbgsmk
> I notice that "My only concern today is heat",which is absolutely correct, and I'll take a warning.

Exactly! 
Sorry, I tried to force myself to write such a nice short comment, but it didn't work. :-)

blackberryer, 2020-12-03:
> The [B012-617DB-1-63-B331](https://github.com/jankae/VNA2/files/5618474/B012-617DB-1-63-B331.pdf) is actually a replacement for the TCM1-63AX+, so yes, that would definitely work as well. The [...]

Oh,thank you for your reply,
then
For tcm1-63ax +, the third winding should also be removed?
[image]

jankae, 2020-12-03:
Yes, you need to remove that winding, otherwise the directivity is really bad at higher frequencies and you won't be able to make accurate return loss measurements.

blackberryer, 2020-12-03:
> Yes, you need to remove that winding, otherwise the directivity is really bad at higher frequencies and you won't be able to make accurate return loss measurements.

Thanks,and I guess it is due to balun's unbalunce?It seems that the balun is not designed for high directionality.

jankae, 2020-12-04:
I believe the issue is that all these baluns are designed for single-ended to differential conversion (I guess that is also implied by the name balun :D). What really is required at that point in the schematic is a common-mode choke. Something like a CX2156NL would be more fitting. However, I couldn't find anything (at least nothing affordable) with a wide enough frequency range.
So removing this tertiary winding is somewhat of a hack to make the wideband (and cheap) baluns work for my use case.

fickybabs, 2020-12-07:
Hi All,

Where can I get all the parts in the BOM? I tried JLCPCB for SMT Assembly, but almost half of the parts are not available, especially the ICs. I've decided to order just the PCB & Stencil and do the assembly old-fashioned way (by hand!). Still, I couldn't find all the parts needed in Mouser or Digit Key.

jankae, 2020-12-08:
Hi, there are some parts that are hard to find, the biggest problem is most likely the balun. As stated above, you can use the TCM1-63ax+ instead (might have to adjust the shielding a tiny bit because this balun is slightly higher).

I didn't do a thorough check but most other parts should be available at mouser as well as digikey (at least they were when I last ordered there for this project). Two exceptions I know of:
* DSB221SDN (TCXO): This is a standard 2.5x2mm 26MHz TCXO, you should be able to find a replacement at mouser/digikey
* C668624 (USB-C jack): Also standard (16 pin) USB-C connector, there are also some replacements available, just make sure you check the footprint carefully.

nbgsmk, 2020-12-08:
Hi @fickybabs !
I share with you my version of BOM, where I tried to compare some of the largest distributors, availability and prices. I checked this last time in November, so expect some changes. It is not absolutely complete, some exotic parts remain to be further investigated, but most are there with catalogue numbers and prices. Sorry the LCSC part numbers are not there. And please compare somehow with Jan's latest BOM as I didn't pay attention if there were any changes since then.
Obviously I have to put a small disclaimer now: I **did not** complete my VNA according to my own BOM! In the meantime, I contacted Mr Hugen and obtained a finished unit from him. Thanks again for his very kind cooperation, and helping me to take the easy way. And shame on me :-)

Back to the actual build, my idea was to use LCSC and their smt assembly service for almost all that is available. I would rather pay, say...10$ (forgot how much it was) than spend two days fiddling with a bunch of resistors and capacitors. If you divide the cost of smt assembly with the working hours it would take you ...is it really worth it? It depends. ;-) 
But! There are several "extended parts" according to LCSC, which will cost an additional 3-4$ each (for mounting the smt reel, handling the machine etc). One by one, they add up to a lot in the end, so I was going to use LCSC smt assembly service for all 'common' parts and maybe the fpga, no more. The remaining 'extended' parts, or otherwise unavailable at LCSC, I was going to order separately from Mouser or somewhere, and do them myself. In that case, when anything is already soldered on the board, the smt stencil would be useless to me, so I didn't consider it. Finally, if I remember well, I was going to order 5 boards, with 2 boards partly smt assembled, as I described. These are the JLCPCB minimums. And I still thought it was worth it, regarding the small difference in price, possibility that I damage one unit during experimentation, and most of all, the time it would take to solder everything from zero.

I do hope my thinking helps you. 
If you go through with your own build, please let us all know. 
I have not completely given up on building one for myself, just postponed it for another time.

Zoran

[Porudzbina BOM i slicno - to be completed.xlsx](https://github.com/jankae/VNA2/files/5659701/Porudzbina.BOM.i.slicno.-.to.be.completed.xlsx)

fickybabs, 2020-12-08:
> Hi, there are some parts that are hard to find, the biggest problem is most likely the balun. As stated above, you can use the TCM1-63ax+ instead (might have to adjust the shielding a tiny bit [...]

Hi Jan,

Thank you for your prompt response. It seems the TCM1 will be available next week on Mouser, and I just found LDL212PU50R on DigitKey. Things are looking good, still have about 6 parts to search for. I like to take this opportunity to thank you for sharing this project. I have all the other nanoVNAs including the recent Plus4, but OWO and edelswartz123 introduced me to your project. This one would work best for my application (6GHz). I currently use a Copper Mountain VNA that cost $5k and only goes up to 5.4GHz.

fickybabs, 2020-12-08:
> Hi @fickybabs !
> I share with you my version of BOM, where I tried to compare some of the largest distributors, availability and prices. I checked this last time in November, so expect some changes. It is not absolutely complete, some exotic parts remain to be further investigated, but most are there with catalogue numbers and prices. Sorry the LCSC part numbers are not there. And make sure to compare somehow with Jan's latest BOM as I didn't pay attention if there were any changes since then.
> Obviously I have to put a small disclaimer now: I **did not** complete my VNA according to my own BOM! In the meantime, I contacted Mr Hugen and obtained a finished unit from him. Thanks again for his very kind cooperation, and helping me to take the easy way. And shame on me :-)
> 
> Back to the actual build, my idea was to use LCSC and their smt assembly service for almost all that is available. I would rather pay, say...10$ (forgot how much it was) than spend two days fiddling with a bunch of resistors and capacitors. If you divide the cost of smt assembly with the working hours it would take you ...is it really worth it? It depends. ;-)
> But! There are several "extended parts" according to LCSC, which will cost an additional 3-4$ each (for mounting the smt reel, handling the machine etc). One by one, they add up to a lot in the end, so I was going to use LCSC smt assembly service for all 'common' parts and maybe the fpga, no more. The remaining 'extended' parts, or otherwise unavailable at LCSC, I was going to order separately from Mouser or somewhere, and do them myself. In that case, when anything is already soldered on the board, the smt stencil would be useless to me, so I didn't consider it. Finally, if I remember well, I was going to order 5 boards, with 2 boards partly smt assembled, as I described. These are the JLCPCB minimums. And I still thought it was worth it, regarding the small difference in price, possibility that I damage one unit during experimentation, and most of all, the time it would take to solder everything from zero.
> 
> I do hope my thinking helps you.
> If you go through with your own build, please let us all know: I have not completely given up on building one for myself, just postponed it for another time.
> 
> Zoran
> 
> [Porudzbina BOM i slicno - to be completed.xlsx](https://github.com/jankae/VNA2/files/5659701/Porudzbina.BOM.i.slicno.-.to.be.completed.xlsx)

Thanks for sharing your BOM. I tried to go through the same process as you when ordering from JLCPCB. The minimum order for the PCB is 5 and the assembly is 2. The total came up to about $240 with shipping without any of the ICs and lots of extended parts. I ordered stencil with it but I thought that will be useless since I would have to but all the parts to use the stencil on a new board. So I cancelled the order and decided to order the PCB with stencil only.

blackberryer, 2020-12-09:
> Hi, there are some parts that are hard to find, the biggest problem is most likely the balun. As stated above, you can use the TCM1-63ax+ instead (might have to adjust the shielding a tiny bit [...]

DSB221SDN (TCXO):Replace with 7L26002015 / SMD-2520 
https://atta.szlcsc.com/upload/public/pdf/source/20190801/C337579_A6B609CD5AF72261EA71705C544062A3.pdf

jankae, 2020-12-15:
Sorry for the late reply, somehow I missed this message.
There seem to be a few different datasheets, this for example also allows 3.3V: http://www.mt-system.ru/sites/default/files/dsb321sdn_dsa321sdn.pdf
Not really sure whats going on there... maybe a new part revision without changing the part number?

blackberryer, 2020-12-19:
> Sorry for the late reply, somehow I missed this message.
> There seem to be a few different datasheets, this for example also allows 3.3V: [...]

Thanks，I am projecting the PCB in JLCPCB. I use KICAD. I have made some changes based on yours, such as the package size. Some traces go to the inner layer to shield the entire module without leaving obvious gaps.
[image]
[image]
[image]

sp9bsl, 2020-12-19:
I was thinking exactly in the same direction - closing some chambers and routing signal traces on inner layers. My first thought was decrease the crosstalk by closing the port1/2 cavities near the source switches IC17, IC18.

blackberryer, 2020-12-19:
> I was thinking exactly in the same direction - closing some chambers and routing signal traces on inner layers. My first thought was decrease the crosstalk by closing the port1/2 cavities near the source switches IC17, IC18.
> Good idea with opening the solder mask on opposite layer of QFNs.

I agree with you very much. IC16, IC17 and IC18 have relatively far physical separation. I guess there is not much impact. Therefore, I only switch the traces near T3, IC21, and IC24 to the inner layer, and then close the shielding cavity. In addition, the resistance and capacitor packaging are reduced, the 5V external power is changed to a 12V external power supply and the packaging of relatively easy-to-purchase components is replaced.

jankae, 2020-12-23:
> I am projecting the PCB in JLCPCB. I use KICAD

Could you share a few details about the conversion to KiCad and how you edited the soldermask for the shielding? I am an absolute beginner in KiCad but would like to move this project away from Eagle eventually. However, I couldn't figure out how to edit/create the soldermask layer with round features. See also #1 and #7 for more details.

About the isolation: I am curious to see how your changes improved that, please share some measurements once you finished the assembly. I am currently attempting to add RF gaskets (as suggested [here](https://github.com/jankae/VNA2/issues/2#issuecomment-732785132)) to improve the isolation. The grooves are already milled but measurements will have to wait until after the christmas holidays.

blackberryer, 2020-12-24:
> Could you share a few details about the conversion to KiCad and how you edited the soldermask for the shielding? I am an absolute beginner in KiCad but would like to move this project away from [...]
Regarding the solder mask, I am also a very stupid method, filling it little by little
like this:
[image]
About the isolation: I routed some traces to the inner layer, then closed the shielding cavity, and modified the 3D file to embed the PCB. The SMA openings were also inserted as much as possible, and then copper skin or the mentioned RF gaskets were used to make the shielding cavity full Possible good contacts, and I will try to stick a ferrite material on the inside.

I will update my measurement data after finished the assembly！

blackberryer, 2020-12-26:
> Could you share a few details about the conversion to KiCad and how you edited the soldermask for the shielding? I am an absolute beginner in KiCad but would like to move this project away from [...]

I have soldered the PCBA, but some problems cannot be solved. STM32 causes the +3v3 power supply of the LM3370 to be only 1.8V, but I remove the STM32 from the board, and the +3v3 power supply normally outputs 3.3v. This means that STM32 cannot work normally, which makes me unable to program it. Can you give me a hint?

jankae, 2020-12-26:
Can you share the schematic? I don't think I can help without it, since you changed the power supply section on your PCB

blackberryer, 2020-12-26:
> Can you share the schematic? I don't think I can help without it, since you changed the power supply section on your PCB

I use the USB power,here is my schematic：
[VNA2.pdf](https://github.com/jankae/VNA2/files/5743585/VNA2.pdf)

blackberryer, 2020-12-26:
> Can you share the schematic? I don't think I can help without it, since you changed the power supply section on your PCB

Oh no,I see,STM32G431CBT6 and STM32G431CBU6 are not pin to pin

jankae, 2020-12-26:
Sorry to hear that... At first I thought maybe you are able to fix it with botch wires, but the LQFP variant is missing a few pins (PC4/PC6/PC10/PC11). You could do without PC10/PC11 (just no debug output) but PC4 is used for I2C. It still might be possible if you find another free SCL-pin to replace it but this starts to get complicated and would also need a software change...
 
Maybe get the other STM variant and solder the UFQFPN onto the LQFP footprint? I'm not sure whether that is even possible but if I remember correctly, at least the pitch is the same

blackberryer, 2020-12-27:
> Sorry to hear that... At first I thought maybe you are able to fix it with botch wires, but the LQFP variant is missing a few pins (PC4/PC6/PC10/PC11). You could do without PC10/PC11 (just no debug [...]

Yes, when I modified the PCB at the beginning, I was compatible with both packages, but later found that the U6 version was difficult to buy, so I only bought the T6 version. 
The reason is that I did not carefully compare the data sheet, I didn’t expect this result, but many things were not smooth sailing.
I have Re-ordered STM32G431CBU6 and updated again after waiting for receipt.

blackberryer, 2020-12-29:
> Sorry to hear that... At first I thought maybe you are able to fix it with botch wires, but the LQFP variant is missing a few pins (PC4/PC6/PC10/PC11). You could do without PC10/PC11 (just no debug [...]

UFQFPN's pin is not long enough to reach LQFP. In the end, I still use fly wire to link to it. But now there is a problem in flashing. I'm not familiar with it. Can you help me analyze my error.
[image]

jankae, 2020-12-29:
I don't have access to my ST-Link right now, but I think it also shows that error. Not really sure what it is about but the important thing is the ** Verified OK ** some lines below. Looks to me as if the flashing worked. Do any of the LEDs light up?

blackberryer, 2020-12-29:
> I don't have access to my ST-Link right now, but I think it also shows that error. Not really sure what it is about but the important thing is the ** Verified OK ** some lines below. Looks to me as if the flashing worked. Do any of the LEDs light up?

LED is blinking GREEN/RED，and LED is RED in the end
[image]

jankae, 2020-12-29:
Sorry, I meant LEDs on the VNA. If you just flashed the firmware, the error LED should report a "Flash empty" error (see also the [user manual](https://github.com/jankae/VNA2/blob/master/Documentation/UserManual/manual.pdf) and https://github.com/jankae/VNA2/blob/master/Documentation/DeveloperInfo/BuildAndFlash.md (Getting a new board running at the bottom of the page).
You can also check the TX pin next to the SWD pins, it should print some messages (115200 baud). If any other LED than the green power LED lights up or you see any debug message on the UART, flashing did work.

blackberryer, 2020-12-29:
> Sorry, I meant LEDs on the VNA. If you just flashed the firmware, the error LED should report a "Flash empty" error (see also the [user [...]

I'm very sorry to disturb you.
The LED on VNA doesn't light up, and the device manager on PC displays DFU device, while I can see the name of VNA device when debugging
[image]

Therefore, I think the hex should not be flashed successfully

jankae, 2020-12-29:
Ok, that does sound like flashing didn't work. Which really confuses me, because the verification appears to be correct. Another strange thing: two lines above the error the STM32**L**4xx is mentioned. Maybe just a typo in OpenOCD?

Two more wild guesses:
* Could it be that you are using the embedded ST-link on a Nucleo board and flashing the Nucleo MCU by mistake?
* Did you do anything with the BOOT0 pin? According to AN2606, the STM32G4 does not start the bootloader with DFU mode automatically when the FLASH is empty, only when the BOOT0 pin is high (or after changing some configuration bits)

blackberryer, 2020-12-29:
> Ok, that does sound like flashing didn't work. Which really confuses me, because the verification appears to be correct. Another strange thing: two lines above the error the STM32**L**4xx is mentioned. Maybe just a typo in OpenOCD?
> 
> Two more wild guesses:
> 
> * Could it be that you are using the embedded ST-link on a Nucleo board and flashing the Nucleo MCU by mistake?
> * Did you do anything with the BOOT0 pin? According to AN2606, the STM32G4 does not start the bootloader with DFU mode automatically when the FLASH is empty, only when the BOOT0 pin is high (or after changing some configuration bits)
> 
> Edit: you managed to debug the program and it already showed up as VNA on USB? During debug, the MCU is already programmed and a simple reset after debugging should restart the program. The fact that you land in the DFU mode afterwards makes me even more suspicious of the BOOT0 pin
Yes, there is another stm32l4 on my st-link. I use the STM32 st-link utility to see that it seems to be burned in, but the FPGA does not light up the LED. Maybe the stm32g431 is not in good contact with the FPGA, as mentioned above “UFQFPN's pin is not long enough to reach LQFP. In the end, I still use fly wire to link to it.”

[image]

I'll check my connection further, anyway, It reminds me: The fact that you land in the DFU mode afterwards makes me even more suspicious of the BOOT0 pin

blackberryer, 2020-12-29:
> > Ok, that does sound like flashing didn't work. Which really confuses me, because the verification appears to be correct. Another strange thing: two lines above the error the STM32**L**4xx is mentioned. Maybe just a typo in OpenOCD?
> > Two more wild guesses:
> > 
> > * Could it be that you are using the embedded ST-link on a Nucleo board and flashing the Nucleo MCU by mistake?
> > * Did you do anything with the BOOT0 pin? According to AN2606, the STM32G4 does not start the bootloader with DFU mode automatically when the FLASH is empty, only when the BOOT0 pin is high (or after changing some configuration bits)
> > 
> > Edit: you managed to debug the program and it already showed up as VNA on USB? During debug, the MCU is already programmed and a simple reset after debugging should restart the program. The fact that you land in the DFU mode afterwards makes me even more suspicious of the BOOT0 pin
> > Yes, there is another stm32l4 on my st-link. I use the STM32 st-link utility to see that it seems to be burned in,
> 
> [image]
> 
> I'll check my connection further, anyway, It reminds me: The fact that you land in the DFU mode afterwards makes me even more suspicious of the BOOT0 pin

I use the DfuSeDemo leaver DFU mode, and the led starts flashing
[image]
[image]

but, When I generate the FPGA bitstream, and assembling the combined VNA firmware, The following error was prompted:
[image]
I don't know why?

jankae, 2020-12-29:
Did you figure out what caused it to boot into DFU mode or do you have to exit DFU mode every time now?
The timeout problem in your last screenshot is between the MCU and the external FLASH (W25Q16). The FLASH has a busy bit which is not reset in time (should only take 150ms typical according to datasheet).
You mentioned that the LED starts flashing, you should get an error code 2 (LED flashes two times, then a short pause, LED flashes two times again and so on). Is that what happens? If you get error code 1 instead, there is something wrong with the connection to the FLASH which would explain the timeout you are experiencing

blackberryer, 2020-12-30:
> Did you figure out what caused it to boot into DFU mode or do you have to exit DFU mode every time now?
> The timeout problem in your last screenshot is between the MCU and the external FLASH [...]
Now it won't enter DFU mode, because one of the pins on this side doesn't contact well, but I can't determine which pin it is. At present, there are still problems in communication with FPGA. I have played the adapter board of ufqfpn to LQFP, and I will further check my flying line before the board comes back.
[image]

blackberryer, 2021-01-05:
> Did you figure out what caused it to boot into DFU mode or do you have to exit DFU mode every time now?
> The timeout problem in your last screenshot is between the MCU and the external FLASH [...]
It worked!!!!
Initial test interface without aluminum shell
[image]

blackberryer, 2021-01-15:
[VNA2_1MHz-6GHz_measurements.pdf](https://github.com/jankae/VNA2/files/5821835/VNA2_1MHz-6GHz_measurements.pdf)
I tested the capacitance of 1nf with the self-made TRL calibration kit, the error was 11.3%, and there was some resonance. This may be caused by the cable I used (CINCH 415-0029-012). When I switched to a semi-flexible cable, the trace The line resonance will be much better

irbsurf, 2021-05-21:
Hi, just wondering, are you planning to release your kicad source files, I'd quite like to take a look at them?
Thanks,
Andrew

bvernoux, 2021-05-22:
Actual KiCad source files are available in special branch https://github.com/jankae/LibreVNA/tree/kicad it is Work In Progress
- For more details on improvements to do see https://github.com/jankae/LibreVNA/pull/7

irbsurf, 2021-05-22:
Hi, thanks for that, I have downloaded the files and I'm now looking at them in kicad!
Okay about the improvements needed, I will read through those, may have a go at adding them.
Andrew

jankae, 2021-12-04:
closing due to inactivity

closed 2021-12-04 by jankae