jankae/LibreVNA #17: Reference clock widgets - consolidation

nbgsmk, 2020-12-07:
Jan,
what do you think about combining "Reference" widgets like this:
Ref. input with options Int, Ext, Auto
Ref. output with options 10M, 100M, Off
In other words, to remove the checkboxes and leave only dropdown combos with those choices.

Zoran

jankae, 2020-12-07:
Sounds good to me, it saves some space in the toolbar while still showing the same information. Do you want to make the changes?

nbgsmk, 2020-12-07:
Sure, I will try.

nbgsmk, 2020-12-10:
Hello Jan!
While consolidating reference widgets, I found when Ref: external is selected, the whole device locks up and traces stop drawing. At the same time, internal temperatures are stopped with 95/95.deg in logwindow, which is obviously a false reading. Anyway the only way out of this is to disconnect the usb cable or switch (example) to signal generator, enable/disable port 1, and go back to vna. Strange, is it not? :)
I am suspecting the pll's or something are getting locked up due to missing clock, so maybe unable to communicate with fpga. Sorry, I'm just guessing here. :) But still the mcu works on it's own clock (I believe it's using HSI) and communicates with the PC, so we should be able to recover easily. I'm sure you already have a solution in your mind.

Zoran

jankae, 2020-12-10:
Your assumption is correct, if you select an external reference without actually applying a signal at the reference input, both PLLs and the FPGA are missing their clocks. No communication with the FPGA is possible anymore, leading to the locked up device. Changing modes recovers from that condition because it resets the FPGA. This behavior can possibly be improved in the device firmware (e.g. add a timeout to the FPGA communication and preform a reset if necessary). The device will still not work without the reference signal but at least it would return to a normal state when the reference is switched back to internal without the need of a power cycle or mode switch. I'll see what I can come up with :)

jankae, 2020-12-15:
External reference behavior improved: 00f0de4
I mostly added some timeouts to the FPGA communication and try to perform a reset if something goes wrong. This recovers from the locked state (when external reference is selected, but no signal is applied at the reference input). There is a small delay (about 1 second) in automatic reference mode when loosing the external reference before switching back to the internal but I think this is acceptable.

closed 2020-12-15 by jankae

nbgsmk, 2020-12-15:
Hi! Works well, no lockup. Practically I could'n think of anything better than what you did - wait for timeout and try to re-initialize.

My only suggestion is: if device logwindow is closed (like "normal" users will probably do), you don't really know what is going on under the hood. Because my pc application does not switch back automatically (is it supposed to?), I would propose a simple alert window like "no external reference detected, please switch manually...". What do you think?
I'd try to do it myself but not sure if I understand that part of the code well enough.