jankae/LibreVNA #18: Crash in marker widget

nbgsmk, 2020-12-10:
Hello my friend!
Sorry that I am so detailed :-) I really like your creation and I play with it every day. It's inevitable to find issues. I measured almost everything in my house, including chairs, tables, plumbing in the walls....haha I'm joking, sorry :)

Ok, so please check tracemarkermodel.cpp, It was causing a crash when changing any marker from "manual" to anything else. I had a hard time to determine it's about deleting child markers, when there are none existing. At least I believe this is the reason. I added checks to see if the returned vector is empty (this is supposed to be "null' in case of vectors), and it worked well.
But now I only _occassionally_ get a crash when deleting a parent of a delta marker. For example, when I delete a parent marker (ex. marker 4), the delta's 'pointer' changes to next available, ex. marker '3'. When I delete this, or create a new marker, then delete... etc...play with this several times, it sometimes crashes but I can not reproduce it exactly. For sure you understand your code much better than me, and probably have an idea already. Please check. 

One more thing along this route, it would be nice to have a visual indication on which marker is clicked in the tree widget. At the moment, you can't tell which one will be deleted on bDelete. I made it with preference dialog but here I couldn't for some reason. Must be too late in the morning now...
Thanks!
Zoran

jankae, 2020-12-10:
I found a way to replicate the crash with the delta marker: Create a delta marker and any other marker type. Use the other marker as the deltas reference. Then, delete the delta marker first and afterwards the other marker -> crash

I believe the issue is this: When a delta's reference marker is deleted, I need to assign a new reference. This is done through a Qt connections but implemented as a lambda function. Because of that, the connection is not deleted when the delta marker is destroyed. Once the delta's reference marker is deleted as well, the code attempts to find a new reference marker for a delta marker that does not exist anymore.
I moved the reassignment into a proper slot and couldn't get it to crash anymore (connection now gets deleted when the delta marker is destroyed): 79a990a

Please check if it fixes the issue for you as well so I can close this issue.

About the highlight of the selected marker: I couldn't find an immediate solution, not sure why. I will check again later.

nbgsmk, 2020-12-10:
You are very fast, my friend! Give me some time to rest my head and I will try to test this today-tomorrow...

In the meantime, maybe this brings some more light: 
You remember the problem with creating a marker and trying to delete it _without clicking_ on any marker. This chrashed the app because modelindex was invalid when nothing was clicked in the treeview. 
Ok, so this one was somewhat similar: crash occured when any marker _had been clicked_ in the treeview and trying to delete it. After many tries, the name beginRemoveRows() made me suspicious about child markers and I tried to add one and change it to delta. So just changing it's type made it crash.
Maybe that leads you to some conclusion.

By the way, could you explain the left-most column in the widget? 
[image]

Why is this number user-editable in the first place? This does not change the graphics of the marker itself, and also if you have, say, 5 markers, then delete number 2 and 3, you will have 1, 4, 5. I understand changing indexes on the fly is not for the weak-hearted people :) But then if you add one more marker, you have markers 1, 4, 5, 2. For me as a self-proclaimed programmer :) it was kind'a confusing as I would expect 6 in any case. And I wonder if this may be causing some disorder deeper  inside? Your thoughts?

nbgsmk, 2020-12-13:
Hello Jan!
I see you have been busy over the weekend. Just pulled the latest commit and it works really nice. I couldn't practically provoke a crash, after a lot of fiddling. And I tried.. :-) I even added delta, then delta-to-delta, then delta-to... :) markers, and it worked great!
Oh, and the highlight on the marker list is really helpful, thanks! ;-)
No crashes here. Shall we close the issue?

jankae, 2020-12-14:
Thanks for confirming, I will close this now.
> Why is this number user-editable in the first place?

I agree, it doesn't really make sense. My initial idea was to allow (alphanumeric) names for markers instead of just numbers, I guess it is a leftover from that attempt. When a new marker is created it always gets the lowest "free" number. There is also another problematic situation: you can give two markers the same number, that could cause some issues when assigning delta markers (not an issue for the code, internally the assignment is handled with pointers. But for the user it would cause some confusion).

But I think I will leave the number editable for now, it doesn't break anything important (and if someone doesn't like it, they can simply leave them at the default number)

closed 2020-12-14 by jankae