jankae/LibreVNA #21: Marker list goes blank occasionally

nbgsmk, 2020-12-15:
Hello Jan!
One thing I noticed, sometimes when changing marker type, the marker list goes blank. Then I press add or delete button and the list is refreshed and all markers visible again. This only happens occassionally, most often when changing type from manual to peak table. Tried to find out the cause but I couldn't be sure if it's my pc graphics or a real code issue.
Zoran

jankae, 2020-12-15:
Definitely a real code issue, it happened quite a lot when I tested it just now. I believe I have found the problem: a call to beginRemoveRows without the corresponding endRemoveRows when helperMarkers are deleted (happens regularly with the peak table, never with a manual marker). Please check if f0c5962 fixes the issue for you as well.

I am not sure why this worked before, I haven't changed that part of the code in a while and was able to use peak table markers without problems before. Maybe other code (that has been changed recently) hid the problem.

nbgsmk, 2020-12-15:
It seems to work just fine. No more invisible markers :) 

One question, though: when low/high/bandpass marker is selected on S11, the left column shows "calculation not possible..." but you can still open child markers and read some data off them. It seems to select an unclear (to me?) value. I don't care much about this, since it's clearly stated data should be disregarded. But honestly I am not sure if this is the same behaviour as before? Whatever you decide to do about this second question, the initial problem seems ok now, so we may close.
Zoran

jankae, 2020-12-15:
Thanks, glad it worked :)
I always created the child markers, regardless of the type of measurement (reflection/transmission). That doesn't really make sense, I changed it just now. A low/high/bandpass marker on a reflection measurement trace will not have any child markers anymore.

closed 2020-12-15 by jankae

nbgsmk, 2020-12-15:
Honestly, it didn't bother me much. As you see, I never noticed it until today.  :)

nbgsmk, 2020-12-15:
Speaking of peak table markers, does it make sense to have also a "minimum table" entry? The opposite of peak table - to find minimums below a threshold. I have never needed any of those, so I must ask if that would have any purpose in reality?