jankae/LibreVNA #23: JLC only takes gerber or other EDA formats, would you please export one in the project?

taskx6004, 2021-03-11:
Hi,

Thanks for your effort and making it open source. After about a year's free time search online RF projects I found your design is the most doable challenge to begin my journey in RF hardware.

I am very interested in making a small batch of PCBA from JLCPCB. I have previous experience with them and it was successful. After checking the JLC SMT parts catalog, they have the specific FPGA and STM32 chips and most RCL parts, but other ICs I will try a hot gun soldering after most passive parts assembled.

Would you please add a gerber folder and put the gerber files for the PCB? I have read previous talks about the stacking, and I will place the spec accordingly when I place the order.

Something else not sure about the BOM is the "100n, 0402, broadband capacitor (e.g. ATC 530L series: ATC 530104KT16T)". could I just use 50v MLCC from Samsung? (with JLC part# C1525), or any other suggestions or a mouser part# please. I can contact JLC with a specific part if available from mouser in China.

One more thing in this issue is the CNC closure, I can design a 3 parts assembly with a from panel to improve SMC socket grounding contact. Will this effort be worth it? 

Thanks.

jankae, 2021-03-11:
Hi,
here are the Gerber files of the current hardware version:
[VNA_2021-03-11.zip](https://github.com/jankae/VNA2/files/6124636/VNA_2021-03-11.zip)
They are nearly identical with the "Second Prototype"-Version described [here](https://github.com/jankae/VNA2/blob/master/Documentation/DeveloperInfo/VersionsAndModifications.md) except that the known mistakes have been fixed (additional inductor at amplifier, pinout of DC jack corrected).

Also, this is the generated placement file and BOM which is required for the JLCPCB assembly process:
[VNA_JLCPCB_parts.zip](https://github.com/jankae/VNA2/files/6124642/VNA_JLCPCB_parts.zip)
Please note that I never used the assembly process with these files, you'll have to carefully check them (especially the rotation of components could be wrong).

> Something else not sure about the BOM is the "100n, 0402, broadband capacitor (e.g. ATC 530L series: ATC 530104KT16T)". could I just use 50v MLCC from Samsung?

That  should work as well. The capacitor I used is specifically specified for high frequencies (and subsequently really expensive) but I actually didn't notice any difference when I used ordinary 100n capacitors I had lying around.

> One more thing in this issue is the CNC closure, I can design a 3 parts assembly with a from panel to improve SMC socket grounding contact. Will this effort be worth it?

I am not sure. Some people reported issues with the contact of the SMA connector (e.g. [changes in the measurement data when applying pressure to it](https://github.com/jankae/VNA2/issues/2#issuecomment-732785132) or [improvement of isolation when adding RF gaskets between the connector and the enclosure](https://github.com/jankae/VNA2/issues/2#issuecomment-751925450)). So far, I couldn't replicate any of those, it might vary from sample to sample. As long as you make sure that you have good contact between the 3 parts of your assembly, I believe it can only improve grounding contact. Whether it is worth the effort you'll have to decide for yourself.

taskx6004, 2021-03-11:
Your response is much faster than I expected. I will start checking process and report back the progress.

The probe socket (SMA connector) grounding issue makes me thinking about a nice video from a youtuber, who replicated a [CERN designed open source voltage-meter](https://www.youtube.com/watch?v=D28uSzCs7-k&t=242s), where a fancy probe grouding was made, I will evaluate the design they used and hopefully bring back something might be useful.

jankae, 2021-04-17:
Just closing up some stale issues, did you manage to get it produced with the files? If everything worked, I'd like to close this as well

taskx6004, 2021-04-18:
Hi, some chips are currently waiting for delivery, sorry about the delay. You can close it now as you wish, please check this thread some time when I have further questions.

taskx6004, 2021-04-26:
Hello,
After waiting for the chip over 4 weeks still not located a complete source of BOM, meanwhile I had good look into the fundamentals of the design of a VNA, I needed a wider bandwidth for my application, so I get started to design a LO+Mixer+ADC board. I am sorry for abandoning my previous idea and your help.

This new board reuses most of the RF circuit from another open source VNA on github, I guess you would know it's from github.com/ttl. My plan is to begin from building a standalone Rx part, then look into Artix7 pcie driver, and a raspberry pi 4 or jetson nx backend to host the software part. I have almost finished the schematics, the final things needed for this rx board is locating a proper priced tcxo, power chips.

I am looking forward to discuss this receiver design in the forum of this project, when the schematic and BOM is finally finished.

closed 2021-06-28 by jankae