jankae/LibreVNA #24: Software (PC_Application) build fails due to libusb-1.0 dependencies in QT5 environment

alextsure, 2021-03-17:
Hello Jan,
Thank You very much for sharing this great project,
Could You, please, give some comments about details of QT5 Development environment setup, especially how should the libusb-1.0 files be installed in the QT5 ?
I have encountered a following problem while trying to compile the PC software in QT5.12.10 (using the MinGW73_64 compiler) under Windows x64: 
"./../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lusb-1.0
collect2.exe: error: ld returned 1 exit status
mingw32-make[1]: *** [Makefile.Debug:581: debug/Application.exe] Error "

It's not obvious from the compiler error log why did the software build fails due to errors with libusb-1.0 included files installation.
I have tried to follow the recommendations and have copied all the libusb *.a files from Libusb-1.0 archive as the instructions says:
"MinGW/cygwin:
  - Copy libusb.h, from include/libusb-1.0/ to your default include directory,
    and copy the MinGW32/ or MinGW64/ .a files to your default library directory."
May be I am doing something wrong, or have set up the compiler environment paths incorrectly, or have missed a '-lusb-1.0' linker option when compiling ?
Any advice would be appreciated, sorry for a messy questions. 
Thanks!

nbgsmk, 2021-03-19:
Hello! 
I was able to set it up like this:
Qt is installed in C:\Qt 
Below left you see where my libusb is, and it's contents. On the right side I marked the two lines which I add to the Application.pro. Surely there is a more correct way, but once I got it working, I was afraid to touch it again. :-)

[image]

alextsure, 2021-03-22:
Thanks for help, 
after substitution of the local path to the libusb files the build goes without errors.

jankae, 2021-03-22:
Glad to hear that :)

closed 2021-03-22 by jankae