jankae/LibreVNA #25: Trace export as png, jpg or similar

nbgsmk, 2021-03-21:
Hello Jan!
I was away from the project for a couple of months, for having to pursue some work related issue. Sorry for that.
In the meantime, I see you have done *a lot* of work and I admire what you have done. I have used the VNA many times during that time, to make measurements for my friends who are not into VNAs' and stuff, or just not so technically-involved. One requirement that come up almost every time is, I had to provide the completed measurements in a form of a screenshot. 
For someone who does not own the instrument, and has no need to install the supporting PC application, this is really the only option to see the results. I had to make a creenshot each time, depending on my desktop or laptop size, or if the app was maximized or not. So kinda' variable, but still very usefull. Can not deny that. :)
So I was wondering, if it was feasible to:
- add a png (or jpg) export option? An additional checkmark "also export as png" during the "Touchstone export" dialog would be more than sufficient. 
In other words, I would save and keep the touchstone export in any case (might be needed much later), even if I would send just the png to someone else.
- Just now it occured to me how a touchstone file contains all four measurements which could not be logically put into a single png image. 
I propose the following shortcut here: provide a single "png" checkmark in the "touchstone" dialog. During the final save-as dialog, you would enter a desired filename and four additional png files like "filename_S11.png", "filename_S12.png" etc would be generated each time. That seems completely functional to me, and still less impacting to the code than adding four separate checkmarks next to each S11, S12... If some traces are not used, it's no big deal to just ignore the png files later.
- Finally, the exported png, according to my imagination (again :-) ), would be say...3000x4000pix, or anything you consider large enough for inspection by the end user. Since we are dealing with math under the hood, the png can be small or big you imagine, without getting pixelated. The end user can later zoom in/out to a large degree, and not be limited by the resolution of the screen where the measurements were exported. In a few years, today's full hd resolution will probably look ridiculously small. :)

This seems like a nice challenge, except in this case I would not even know where to start, from your code.
If you were doing only this project every day, you would still be very very busy. Therefore, this is just an idea, don't take it too seriously. :-)

Regards,
Zoran

jankae, 2021-03-22:
Hi Zoran, welcome back!
At the moment I don't feel like the effort will be worth it:
* It is not straight forward to export a png, there would need to be a lot of user configurable parameters (comparable to the graphs I currently have, and they are still pretty limited compared to other programs)
* There is more than one way do display a touchstone file (e.g. smith chart, ...)
* Free touchstone viewers are available that already do a much better job than I ever could (e.g. http://magimaji.com/). I believe using the raw data (the touchstone file) and such a viewer is the best solution at the moment

nbgsmk, 2021-03-22:
Ok Jan, thanks! The idea was to "print" the png just as it is currently displayed on the screen (be it xy or smith), possibly in some higher resolution. I was primarily thinking of exporting what I use most often, and that is either a single S11 or S21 graph on the screen. But you are right: how to display the other graphs, if they are not currently visible in the pc application (smith, xy, which scale...)?  I begin to realize several hidden issues exist. But no big deal, it was just a question.
Regards,
Zoran

closed 2021-03-22 by nbgsmk

jankae, 2021-04-12:
With e66fc01 this is sort of implemented now (save png as seen on screen, no higher resolution). Accessible either via File->Save image (this saves a screenshot of all graphs) or by right-clicking a specific graph and selecting Save image (saves image of only the selected graph)

nbgsmk, 2021-04-17:
Works like a charm! 👍