jankae/LibreVNA #29: Software build / toolchain setup - STM32

nbgsmk, 2021-04-25:
This is another very low priority "issue."

Now that I took a dive into the embedded part of the VNA, building the stm32 code will obviously have it's benefits.
I am using the latest STM32CubeIDE, which is 1.6.1.

- I just tried, once more, to:
create a new workspace in eclipse
download a zip archive of the project
"import existing projects into workspace" (choosing the mentioned zip)

Ok sofar, except a tiny little issue :)   Using ctrl-B (build all), I get the message 
"Build of configuration Release for project VNA_embedded
Nothing to build for project VNA_embedded"

- Regarding the .ioc file, it opens after this message
[image]
but regardless of continue or migrate, and generating the code from the wizard (or skipping this step) the problem remains "nothing to build"
And there is not much to do in the .ioc file, anyway. :) 

As I am already handling the git aspect using QtCreator, I need STMCubeIde to just pick up the code already sitting on my disk. In any case, build should have nothing to do with the project being imported from git, or taken from a zip.

Help is appreciated, but again, treat this as *VERY LOW* priority. 
I'm sure you have better things to do and, even though I am always enthusiastic to fiddle with stm32, I don't see myself giving much contribution, as the code looks quite compact and complex. On the other hand, I expected even less in Qt, yet here we are... :) :)

jankae, 2021-04-26:
I am using a slightly different setup, namely the combination of CubeMX and SW4STM32. I believe the STM32CubeIDE is essentially the same thing, just merged into one program.

I just downloaded it (also version 1.6.1) and am having the same issue after File->Import->Existing Projects into Workspace (this normally works for me on SW4STM32 when moving a project to another computer).

However, File->Import->Import ac6 System Workbench for STM32 Project worked. I had to select the correct MCU (STM32G431CBUx) and there was a message about "converting" the project for the STM32CubeIDE (whatever that means). Afterwards I could successfully compile the project without a problem. Maybe that works for you as well?

alextsure, 2021-04-27:
Hello Jan,
I have also tried to import the STM32 Project into the  STM32CubeIDE (just for the testing purposes), and it worked OK with several minor issues reflected in the logfile.
After the project import, I tried to build it in STM32CubeIDE  with the default settings.
The build in "Debug" configuration run without errors, but the size of the output *.elf file is more than 2 Mb, possibly, due to some extra debug code added into the output file by the compiler (?).
At the same time, an attempt to build the same project in the "Release" configuration failed with the following errors:

../Src/main.c:793: undefined reference to `App_Start'
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:78: VNA_embedded.elf] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

line 793 in main.c contains the empty function
 /* USER CODE BEGIN 5 */
  App_Start();

Thanks in advance for any comments on this :)

jankae, 2021-04-27:
> but the size of the output *.elf file is more than 2 Mb, possibly, due to some extra debug code added into the output file by the compiler (?).

Yes, the *.elf file contains a lot of extra data but you can still flash it into the MCU (the flashing tool will extract the necessary bytes). Unless you want to debug the code, you will find the *.bin file more useful as that is whats used together with the FPGA bitstream to create the VNA firmware. The *.bin file should be below 128kB, otherwise it will not fit.

> At the same time, an attempt to build the same project in the "Release" configuration failed with the following errors

Huh, looks like I had the settings for the release build setup wrong (not compiling the files in the Application folder). I only ever used the debug build (the github actions build a release firmware but they are using a separate makefile because I couldn't get them to work with an IDE). This should be fixed with the latest commit, please try importing the project again.

closed 2021-10-09 by jankae