jankae/LibreVNA #32: New Feature Suggestion - 2-port TDR

tim292stro, 2021-06-24:
I'd be interested on thoughts for enabling Time Domain Reflectometry post-processing - probably wont see useful bandwidth above 3GHz, but that's probably enough for most people who buy this device.

jankae, 2021-06-25:
Could you explain a bit more what kind of features that would include? A simple TDR calculation is already possible (but it may be a bit hidden):
1. Double click a trace, the "Edit Trace" window opens. Click the "plus" button to add a new math operation:
[image]

2. Select the TDR option and configure it to your needs:
[image]

3. The "Edit Trace" window should show the added math operation and also indicate that this trace will now output time domain data:
[image]

4. As the domain of the trace has changed, it will have been removed from the graphs. Drag-and-drop it onto an XY-Plot to view the time domain data.

For useful results, the start and stop frequency needs to be chosen carefully and it also depends a bit on the selected settings for the TDR.

tim292stro, 2021-06-25:
Hmm, I'll look at that when I get home.  This may just turn into a UI improvement request if it's already there (but hard to find).