jankae/LibreVNA #33: Electrical Parameter Trace Measurements Z,R,L,C etc

GoodTesla, 2021-06-25:
It would be nice to have built in traces for derived values. Some Examples:
Series R,L,C Quality factor, Z, VSWR, etc.

jankae, 2021-06-27:
I assume you want to measure these parameters by an S11 measurement? E.g. a capacitor to GND would be connected to port 1 and you would like to see its capacitance instead of the S11 value?

jankae, 2021-06-27:
A first attempt at the feature: e27e594
You can find the compiled version here: https://github.com/jankae/LibreVNA/actions/runs/977042194

There are additional axis types available for the graphs as well as different output formats for the markers:
[image]

This is barely tested at all, I just wanted to provide a quick prove of concept. If you find any errors in the calculations please let me know. I will also check it more thoroughly at a later time.

GoodTesla, 2021-06-28:
Wow that was fast! I will give this a try and let you know if I see any issues. 

Great tool! Been following this project for a while and we finally got a unit to play with in house. 

Thanks again.

closed 2021-10-09 by jankae