jankae/LibreVNA #40: New feature suggestion: Power sweep

arielmello, 2021-07-08:
Hi jankae,
I have recently got my LibreVNA and have been using it to characterize and tune a power amplifier in the 2,45 GHz band.
So far everything looks very good with the VNA and have not found any issues, but I have realized that for people working on PAs it is often very illustrating to be able to perform power sweeps to obtain the Pout vs Pin of the PA. I find out that this feature is underestimated on mainstream VNAs. Usually this task involves either programming a signal generator and a 2 channel power meter or of course doing it by hand, which is long and tedious. Also temperature drifts are somehow masked in this process, as the devices would be operating at different temperatures when doing this in such a slow procedure.
I figured that as mostly all of the primitives for controlling the signal generator are there already, I was thinking that adding a graph on the signal generator tab with possibility of tracing the power detected on one port on one axis and the power delivered by the other port would not be a very complex endeavor. It would be a great app for PA testing. Calculating the compression point at 1 dB would be also very helpful, but of course, it is only luxury.
Maybe it has already been programmed and I have not found yet how to do it. If so, could you please explain how to do this?
Thanks for all the great work you have been putting together and good luck with your future projects as well.

Ariel

jankae, 2021-07-08:
Hi Ariel,

you are not the first person to ask for a power sweep but no one else suggested adding it to the signalgenerator mode. I always thought it would have to be in VNA mode and that would mean quite dramatic changes to the data structures, so I didn't feel like taking it on yet.

One thing you have to keep in mind is the limited granularity of the output power. It is controlled by a step attenuator with a step size of 0.25dBm. So the number of points in the sweep has to be limited somewhat (or multiple points will just use the same attenuator setting). What I could imagine is adding a list to the signalgenerator mode. For each entry in the list you could specify the frequency, output power, output port (1 or 2) and the duration of the signal. Maybe the GUI could also have some helperfunctions to quickly populate that list with a sweep in either frequency or power.
Measuring the power at the ports in signalgenerator mode would require some modification to the USB protocol but it should be doable. The most work would probably be the graph for the measurement results. I am not sure whether it is easier/better to extend the already existing graph class with a different X axis (power) or to write a new graph type specifically for the signalgenerator mode. Extending the VNA/SA graphs could potentially be useful if I ever add a power sweep functionality to the VNA mode...

Maybe I can add the list to the signalgenerator soon(ish) but I believe it will be some time before any useful power sweep with readback could be available. At the moment, the only way you can do the power sweep is by scripting the functionality with SCPI commands. You could set the VNA mode to a short span with just three points (I believe just one point could cause some issues). Then simply setup the different output powers one by one and get the middle point of the sweep for each of them. It might be some work to set up initially but then it should do the measurement quite fast.

Best regards
Jan

arielmello, 2021-07-08:
Hi Jan, 
Thanks a lot for the prompt reply and the suggestions.
I reckon this would require a lot of effort put into. But it is an improvement proposition. Keep it in the back of your mind for the future. I will investigate about the scripting functionality and eventually check out in the future for the signal generator list. Thanks for the idea.

Other than that, I think that the granularity of the attenuator step is not a problem. 0,25 dB is a decent resolution to obtain a compression curve. 

Keep up the good work :)
Cheers,

Ariel

sp9bsl, 2021-07-08:
Hi Jan, Ariel,
this would be very expected feature - keep fingers crossed for it ;)

Slawek

jankae, 2021-07-09:
Well, I thought some more about it and now think it is not as complicated to add a power sweep the VNA mode as I initially believed. With 6748908 I got most of the basic functionality working:
* power as an additional X-axis option for the graphs
* a choice between frequency and power sweeps

This required some changes both in the USB and the FPGA protocol, so a firmware update on the device is definitely required. If you are curious you can have a go at this version but keep in mind that it is very much a work in progress. Basic setup should be fairly self-explanatory:
* Select power sweep in the sweep toolbar, set the frequency and the start/stop power level
* When changing the sweep type, the traces vanish from the XY-plots (because the data domain changes from frequency to power, you have to drag-n-drop them again. This could probably be improved in future versions)

arielmello, 2021-07-09:
Hi Jan,
Can't wait to take a look at it. Thanks for the big work put here.
Will give you some feedback after testing it.
Cheers,
Ariel

jankae, 2021-07-10:
I have done some more testing on the power sweep and adapted various other stuff to it:
* Markers now work on power sweep traces (with special P1dB marker type to automatically find the compression point)
* The graph behavior when the trace changes domain (frequency, time, power) is configurable (see preferences->graphs). Available options are:
  * Remove the changed trace (same behavior as with previous versions of the GUI)
  * Adapt the graph to the new trace (the new default setting, may remove other traces if they are not compatible with the new graph setup)
  * Only adapt the graph if the changed trace is the only one displayed (otherwise the trace is removed)
* Minor other changes to handle power sweeps (e.g. in custom math expressions)

It seems to work quite well and so far I haven't found any issues (but I am sure that the occasional bug is still hiding somewhere). Please let me know if this new feature works well for you or if you have any ideas for improvements.

arielmello, 2021-07-12:
Hi Jan,
Once again thanks for the keen interest you are putting into developing this feature once more.
I have tried to get started using it, but unfortunately I might not be experienced enough with the interface. This is what I do.
Download the latest software and update the firmware of the unit.
Open the software, leave the central frequency at 2,45 GHz and a span of 200 MHz. Then, apply a SOLT calibration.
Switch into power sweep.
Select the operating frequency at 2,45 GHz, with sweep going from -10 to 0 dBm.
Generate a new trace  with the custom expression (P*x^2) from the S21 overwrite function.

But with this process nothing is shown on the graph. (NaN is shown when adding the trace to the cartesian graph)
What is it that I am not doing correctly?

I figure that an interesting feature in the future and probably not 100% related to the power sweep is the history mode, where a trace can be kept on the screen to be able to compare it with any future traces. I reckon that maybe it is actually possible to do this maybe using the max hold feature for a separate trace, but for ease of use maybe a separate button to hold the current trace in history can be useful. Not sure if you have used ADS, but the idea is to have the same function here.

Once again, thanks
Ariel

arielmello, 2021-07-12:
Ok, so it seems that the auto Y axis adjustment was introducing the issue. Now I have removed the Y axis scaling and manually introduced some values and got 2 traces:

arielmello, 2021-07-12:
[image]
Stimulus represents expression "P"
Power on P2 represents expression"P*x^2"
By the way, I have a through connection between port 1 and port 2.
So I expect a positive slope trace to be present here, with slope +1dBm/dBm but get this decreasing trace.

jankae, 2021-07-12:
I believe most of the issues are caused by the custom expression. The result of the expression is interpreted as a complex voltage vector (same as the 'x' parameter). The 'P' parameter is in dBm. To convert this to a voltage vector, you'll have to use 10^(P/20), this gives you the straight line you are expecting:
[image]
The same thing causes the autoscaling problems. Your expressions evaluates to 0 for P = 0, which corresponds to -infinity dBm and the autoscaling is unable to handle infinity.

>I figure that an interesting feature in the future and probably not 100% related to the power sweep is the history mode, where a trace can be kept on the screen to be able to compare it with any future traces.

You can stop (freeze) a trace. The easiest way to achieve your desired behavior is probably to duplicate the trace first (by right-clicking on it):
[image]
This also copies all math operations. Then you can just stop one of the duplicates by clicking on the 'play' symbol (it will change to 'pause'):
[image]

arielmello, 2021-07-12:
Thanks Jan again.
So indeed, it was my bad taking the P value to be in linear. 
I manage to obtain the straight line as you have shared on your message.

Now I represent the S21 parameter in the power sweep mode.  I have my sweep limits set from -10 dBm and 0 dBm and obtain the following curve:
[image]

I have previously calibrated on the frequency sweep mode using a stimulus power of 0 dBm. So, in the vicinity of 0 dBm on the power sweep I see that the S21 parameter is close to 0 dB, which is correct, as I use a through connection on the VNA. However, for lower powers, the S21 parameter is closer to -2 dB. If I calibrate on the freq sweep using a lower stimulus (i.e. -10 dBm), i get the opposite values (0dB for sub -5dBm and ~+2 dB for 0dBm.) Would this mean that there is some limitation on the values of the sweep? Or that a different calibration is to be used for this?

cheers,
Ariel

jankae, 2021-07-12:
This is almost certainly a saturated amplifier in the reference receiver. At higher power levels, the amplifier can not drive the ADC high enough (gain of that amplifier should probably be a bit lower). I do have ADC level checking that should result in an ADC overload warning in the bottom right corner. But apparently I have set the limits for this warning too high, it is not triggered with your measurement (I just checked with my unit, same behavior). I'll adjust the warning threshold. You still have to limit the output power to approximately -6dBm but at least you should get an indication if this problem is present.
A more permanent solution is to change the gain of that amplifier. If you want to, you can do that yourself, it just involves changing two resistors (R94 and R95). A value of 3k9 instead of 4k7 could get you good results (if you decrease that value too much, you'll raise the noisefloor and lose dynamic range)

arielmello, 2021-07-12:
Hi there again,
Would be happy to look into changing these resistors. Is it R94 and R95 resistors?

arielmello, 2021-07-13:
Hi Jan,
I have put some 22k resistors in parallel with R94 and R95 to reduce the gain of the amp. The equivalent resistance should be close to 3k9.
Here is what I get after changing the gain, calibrating at 0 dBm and performing a power sweep from -15 to 0dBm:
[image]
There has been no substantial change in the performance following the modification. If you have any further propositions, let know.
Anyway, thanks for the help so far, I have been using the sweep for an amplifier tuning and it works wonders under this range. I will be testing shortly the P1dB marker.

Ariel

jankae, 2021-07-13:
Hi Ariel,
you are right, adding a 22k resistor in parallel didn't do much for me as well (it did improve the situation slightly but nothing significant). However, a 4k7 in parallel (2k35 in total) got me this result:
[image]
The variation is now down to about 0.13dB over the sweep. If I would have bothered to do the calculations, I could have given you that value immediately. My guess of 3k9 wasn't that good, sorry about that.

I am not sure what causes the remaining variation (probably some nonlinearity in the mixers/amplifier?) but hopefully this is good enough for you.

arielmello, 2021-07-14:
Hi Jan again,
Can I ask you what settings you are using to calibrate? (center freq, span, level?)
I have replaced the resistors to have a total resistance of 2,35 kOhm as you have proposed. Then calibrated with a through standard at 2,45 GHz span 900 MHz, power -5dBm and when switching to the power sweep (2,45 GHz, -20dB to 0 dBm) with the through standard, I get the same as the previous curves. Can you confirm that you get a better curve under the said conditions? 
I find it strange that on your curve, the S21 parameter is never 0dB (associated with the calibration point). Were you measuring a through standard?  
Maybe my soldering is not good? (i fitted 0603 resistors in the 0402 pads...)

jankae, 2021-07-14:
> I find it strange that on your curve, the S21 parameter is never 0dB (associated with the calibration point). Were you measuring a through standard?

I didn't calibrate at all, that is why my curve is offset from 0dB. For the overall shape of the curve the calibration is not important since the frequency is constant across the sweep (will use the same calibration point for the complete sweep). If I follow your steps (calibrate at 2.45GHz with 900MHz span and -5dBm). I get this result:
[image]
As you can see, the curve now passes through the 0dB point at approximately -5dBm (as expected).

> Maybe my soldering is not good? (i fitted 0603 resistors in the 0402 pads...)

Unless you messed that up really badly, it shouldn't have much of an influence. There are no high frequencies present at that part of the circuit. Just to confirm, you added the resistors here, right?
[image]

Please set up the manual control like this and check the ADC limits at the reference receiver:
[image]

You should see similar values there. When using the same settings on a non-modified sample, I get approximately +/-28000 instead.

arielmello, 2021-07-15:
Hi Jan,
Sorry for taking my time to answer.
The problem was I did not identify correctly the IC. I was soldering the resistor on the other differential amplifier (same part number) associated with the port 2 receiver. Once again, my bad. I am still familiarizing with the unit. In my defense, my unit did not came with silkscreen. ;)
I have now verified that the VNA does not saturate for 0dBm input, and have the same variation as you have shown.  

However, now I am testing one stage of the PA, giving the following curve:
[image]
I believe that the interval -2dB - 0 dB still shows some saturation behavior.
Here is the Pout Vs Pin curve, that shows this small region of inaccuracy.
[image]
I think for my application, it is ok, I have to keep in mind to discard the values in this region. 

I also attach the manual screen values i get during the power sweep, which seem to be different to the ones you have shown. Again, maybe I do not yet understand in which mode I should take the capture.
[image]

I will keep on testing this new feature and keep you informed.
Thanks again for all the effort. It is very much appreciated.

Ariel

jankae, 2021-07-17:
> I believe that the interval -2dB - 0 dB still shows some saturation behavior.

Yes, the straight line above approximately -1.7dBm is the limit of the VNA, at that point the attenuator is already set to its minimum attenuation and higher signal levels are not possible (reaching 0dBm is not possible over the complete frequency range, the signal level drops towards higher frequencies). In the signal generator mode, there already is an "unlevel" error flag showing up in that case. I will add that flag to the VNA mode as well.

Another thing: have you performed a source calibration on your unit? It is required if you want to reach accurate output signal levels. For a lot of VNA applications where you don't really care for the exact signal level you can omit that, but in your case you will get inaccurate results if you don't calibrate the output signal level (see user manual, chapter 6).

> I also attach the manual screen values i get during the power sweep, which seem to be different to the ones you have shown. Again, maybe I do not yet understand in which mode I should take the capture.

You have to set the same settings as in my screenshot. In your example none of the PLLs are activated, so you are not getting any signal at the ADCs. But since you found the mistake already, this is not really required anymore.

jankae, 2021-12-04:
closing due to inactivity

closed 2021-12-04 by jankae

satfan52, 2023-08-26:
Hello, where are we standing on this. Is the LibreVNA hardware and software available for sale in 2023 able to do power sweeps without any modifications or is some cooking required ? What is the dynamic range of the power sweep possible ? Thanks

jankae, 2023-08-26:
Yes, power sweeps are possible. But some limitations apply, e.g. the minimum power step is 0.25 dB, so power across the sweep is changed in discrete steps rather than a continuous adjustment.

The available dynamic range is always about 40 dB, but towards the higher frequencies it is shifted downwards (because of the higher losses). And if you want precise results, you should perform a source calibration first, otherwise the absolute signal level may deviate quite a few dB from the set level.

The GUI supports power sweeps without any limitations (that I am aware of). No need to enter any custom math expressions, just set the sweep type to power.

satfan52, 2023-09-08:
Thank you for your reply. What do you mean by source calibration ? Measure the the output power when the attenuation is minimal (0dB) and enter that value in the GUI ?  Is there the possibility to enter an "offset " in the GUI, for instance to specify in the GUI that 0 dB is corresponding to say -10dBm  ?

jankae, 2023-09-13:
Yes, that is the source calibration. For more details on what exactly you need to perform please check the user manual, chapter 9.1 (Source Calibration)