jankae/LibreVNA #41: New Feature Suggestion - VSWR Circle

itsaname, 2021-07-12:
It would be nice to have built in traces for certain vswr circle,its very useful in antenna match. Some Examples:
vswr=1.5, vswr = 2 etc.

jankae, 2021-07-14:
As it is now, the trace module does only support traces created from the incoming measurements or from touchstone/CSV files. 
Maybe it could be easier to just add the option to display built in lines directly on the graphs? (without actually creating a trace and filling it with data).
What other built in lines/traces do you have in mind?

itsaname, 2021-07-15:
sorry for my poor English, it would be nice to add  Q-arcs too.

jankae, 2021-07-17:
okay, that sounds like it could just be an addition to the smithchart and should be doable. I'll probably implement that at some point, thanks for the suggestion

jankae, 2022-01-07:
Well, it took a while but customizable circles/arc for the smith chart are now available: cd106b3. You can configure them in the Setup-Dialog of the smith chart:
[image]
Available options are VSWR, constant resistance, constant reactance and constant Q. Here is an example how it looks like on the smith chart:
[image]

closed 2022-01-07 by jankae