jankae/LibreVNA #46: New Feature Suggestion - SCPI Touchstone export

spel-oe, 2021-09-07:
directly provide a touchstone export function via the SCPI Interface
so that not a manual export of all traces and a manual conversion into a touchstone file is necessary

jankae, 2021-09-07:
Good idea, I think that should be doable. You would still have to specifiy the traces you want to use for the touchstone file as each trace is handled independently from each other in the GUI (there is no information which traces "belong" together). I am thinking of a command like :VNA:TRACe:TOUCHSTONE \<trace1\> \<trace2\> \<trace3\>...
Depending on how many traces are specified, a touchstone file with the corresponding number of ports will be created (e.g. 2 ports if 4 traces are specified). With the default traces, the command would look like this:
`:VNA:TRACE:TOUCHSTONE S11 S12 S21 S22`
and would return the content of that touchstone file as the response to the SCPI command. If the traces do not match (e.g. different number of points, different frequencies) an error will be returned instead. Does this sound useful to you or did you have something else in mind?

jankae, 2021-09-07:
I went ahead and implemented this feature: 3176037
Please see the [programming guide](https://github.com/jankae/LibreVNA/blob/master/Documentation/UserManual/ProgrammingGuide.pdf) on how to use it (search for touchstone).

spel-oe, 2021-09-08:
Thanks for the great work, I'll try this feature as soon as possible

closed 2021-09-08 by spel-oe

spel-oe, 2021-09-08:
Tested this new feature, is working as expected :)