jankae/LibreVNA #61: Confusing behavior around 1 MHz

bwalker000, 2021-10-20:
I am using the device to measure the complex, frequency dependent impedance of an ultrasound transducer. When I calibrate the system the apparent resonant frequency is at around 1.3 MHz, but when I use the default calibration (uncalibrated) the apparent resonance is around 1 MHz. This matters a huge deal for what I'm doing. I've attached some screen shots.

Note that this transducer was designed for a 1 MHz resonant frequency. I'd expect some variation, but not likely bumped all the way up to 1.3 MHz.

I would expect to see some changes in measured impedance. I would not expect the changes to be as dramatic as they are.

Explanations / insights are welcome.

First image: Measured impedance of the 50 Ohm load, with calibration turned off.

[image]

Second image: Measured impedance of the 50 Ohm load, with calibration turned on.

[image]

Third image: Measured transducer impedance with calibration turned off.

[image]

Fourth image: Measured transducer impedance with calibration turned off.

[image]

Fifth image: Measured transducer impedance from a nanoVNA.

[image]

jankae, 2021-12-04:
I finally got around to check this. I don't have an ultrasound transducer but I built a simple resonant circuit (L+C in parallel) and measured that instead:
[image]
I have used a 3.3nF capacitor with a 6.2uH inductor, so the calculated peak should be at 1.113 MHz. The LibreVNA measures it a bit higher (at 1.14 MHz) but this might also be due to component tolerances.
Measurements without calibration don't give any useful information in this case. For my example, it shows a peak (resonant frequency) at about 200kHz. It might be helpful to look at the smithchart for an explanation: the peak of the resonant frequency happens when the trace in the smithchart crosses the "open" point (or gets close to it). A simple phase error (which will definitely be present when no calibration is used) will shift this point significantly, resulting in a completely different resonant frequency.

You can achieve the same effect by adding a cable after calibrating (it has to be rather long because the frequency is relatively small). In my case, the peak shifts down to 1.12 MHz when I use a 1m SMA cable between the resonant circuit and the LibreVNA (calibrated right at the port).

I'd double check your setup (do the nanoVNA and LibreVNA have the exact same setup?) and maybe try a lower IF bandwidth (I have no experience with ultrasound transducers, if they have a narrow-ish bandwidht that might help). And don't trust any measurement without calibration.

bwalker000, 2021-12-04:
Thanks so much for checking into this. Just yesterday I was in the lab and found another way to test this. Short answer is that the resonance of the transducer is actually at 1.3MHz. I suspect that it has to do with the manufacturer not fully accounting for the material it was mounted on. This (coupled with your assessment) gives me strong confidence that the LibreVNA, when calibrated, is giving a correct result in this frequency range. 

I can no longer compare to the nanoVNA as I bricked it when trying to update the firmware.

I continue to be very impressed by the LibreVNA and your commitment to the project.

Thank you!

closed 2021-12-04 by bwalker000