jankae/LibreVNA #69: SCPI Interface sends LF's on commands that do not return values.

djfarrell, 2021-11-19:
Not sure if this is intended behavior.  The SCPI commands I use to configure the VNA return a LF which must be read or these characters in the rx buffer confuse subsequent commands that do return values. See the octave thread in the group.

I wonder if you could add a contributed section of the project in git.  Things like the octave example may be useful to others.

jankae, 2021-11-19:
To quote the programming guide: "Every command generates a (possibly empty) response, terminated with a newline character."
So yes, it is intended behavior. However, I can't remember exactly why I implemented it like that. Maybe so there is a response and you don't just send commands in the dark? You might be able to convince me to change that, but I would prefer to keep it like it is (so already existing and working scripts aren't broken).

> I wonder if you could add a contributed section of the project in git. Things like the octave example may be useful to others.

I actually meant to ask you about that, I think it would be a good example as well. Feel free to create a pull-request for that, it should go in this folder: https://github.com/jankae/LibreVNA/tree/master/Documentation/UserManual/SCPI_Examples

djfarrell, 2021-11-20:
Not sure it matters too much, it can be confusing in that if you write a number of commands before reading a number of leading LF's will build in the buffer.  I changed to reading every time I write. I also now check that "ERROR" is not returned as it could be. 

I cleaned up the octave code and added ifft() to generate a time domain plot, I'm not quite sure what (if any) extra calibration is need for TDR. I'll post is soon.

closed 2021-11-25 by djfarrell