jankae/LibreVNA #71: New Feature Suggestion - Mixed-Sex Calibration

dsbrantjr, 2021-11-26:
Folks:  I have a Kirkby 85033 calibration kit which has both male and female standards, as do many other kits.  I do not see a way in the LibreVNA software to handle a mixed-sex calibration with Port 1 and Port 2 being of different sexes with either a zero-length or non-zero thru.  Can this be accommodated by the current ".calkit" file structure and calibration algorithms? 73, Don N2VGU

jankae, 2021-12-01:
This is now implemented (55ac50e + 93ad894), I hope it is at least somewhat intuitively to use. The calibration kit has a new option to choose whether standards with different sexes should be defined:
[image]
If set to "separate", the short, open and load standard definitions get a male/female tab with separate values:
[image]
If separate values are defined, the calibration trace dialog has an option to select the gender for the standards at each port:
[image]

This required a change to the calibration kit and calibration file format (which has finally been changed to json). Loading old files still works but obviously no male/female standards are possible with them as only one set of standards is saved there. It is probably a good idea to load and save old calibration files to update them to the new file format.

The usual disclaimer: This addition is not ready for a release yet. It worked in my tests but you can still expect a bug or two. If you don't mind that, please download the development version [here](https://github.com/jankae/LibreVNA/actions/runs/1528098149)

dsbrantjr, 2021-12-02:
This is outstanding work and is just what I had in mind.  Thank you.

Will the new software "know" that if one port is male and the other female that a zero-length thru (direct connection) can be used, or is an actual thru always required?  As I recall some calibration kits do not have a male-female thru.  
Would a radio button be needed to choose "Port 1 connected to port 2 via through standard" and "Port 1 connected to port 2 direct."?  I am not certain how this should be handled but I would prefer to have the option to do either.  

I also have an additional UI suggestion:
On the "Calibration Measurements" section, add another field between "Type" and "Prerequisites" for "Gender" and have its content be populated from the "Port 1 Standards" and "Port 2 Standards" radio buttons.  Then there would be a double-check to help ensure that a conflicting standard/definition will not be used.  
Then the entries would read "Port 1 short | Male | Short standard connected...." which would prompt the operator to select the correct standard.

dsbrantjr, 2021-12-02:
Thinking about it further, if we are going to allow a zero-length (direct-connection) thru we will need a way for the instrument to know whether it needs to use the thru standard definitions or set the coefficients to zero.

jankae, 2021-12-02:
Yes, you are right, an option for a zero-length through is also required. I have added that now:
[image]
It is only available if different sexes have been selected for port 1 and port 2, otherwise it defaults to the through standard defined in the calibration kit.

> I also have an additional UI suggestion:
On the "Calibration Measurements" section, add another field between "Type" and "Prerequisites" for "Gender" and have its content be populated from the "Port 1 Standards" and "Port 2 Standards" radio buttons. Then there would be a double-check to help ensure that a conflicting standard/definition will not be used.

Good point, I will add that in the future.

jankae, 2021-12-04:
> Then the entries would read "Port 1 short | Male | Short standard connected...." which would prompt the operator to select the correct standard.

... this is now implemented as well

closed 2021-12-06 by jankae