jankae/LibreVNA #72: New Feature Suggestion - Unkown Thru Calibration Standard

dsbrantjr, 2021-11-27:
The "unknown thru" calibration technique offered by Keysight is powerful and convenient and can allow accurate measurement of DUTs with non-insertable connector pairs or even different connector types, without clumsy adapter characterization and de-embedding steps.  
This would be an ideal feature addition if the VNA architecture allows.

jankae, 2021-12-04:
I am not absolutely sure but I believe this procedure would require a hardware with 4 receivers. At least it looks like that to me, I haven't gone through the math completely yet. So unfortunately, this can not be implemented with the LibreVNA

jankae, 2022-01-07:
closing for now - if I am wrong and someone points me to the math for unknown thru calibration with three receivers, I'll open it again

closed 2022-01-07 by jankae