jankae/LibreVNA #73: New feature suggestion

Chorbu, 2021-12-04:
Hello all,

May is it possible to add an option in Preferences / graphs to increase or decrease the thickness of traces ?
and add a button in calibrations measurements to erase all calibrations in only one action ?
May is it possible when adding a marker in S21 trace for example that others automatically set in the smith chart or other traces ?

Best regards to all
Thank you in advance Patrick F5SUE

jankae, 2021-12-05:
Hi Patrick,
> May is it possible to add an option in Preferences / graphs to increase or decrease the thickness of traces ?

Yes, possible and added now:
[image]

> and add a button in calibrations measurements to erase all calibrations in only one action ?

You can already select multiple measurements at the same time (using Ctrl/Shift) to measure them at the same time (obviously only possible if they do not require conflicting calibration standards). With the latest commit, you can now also delete all selected measurements at once.

> May is it possible when adding a marker in S21 trace for example that others automatically set in the smith chart or other traces ?

This is a tricky one. Markers are linked to one trace, not a set of traces. This has been discussed before and while it does impose some restrictions, it also provides a huge flexibility when importing new traces or deleting some. I assume you want to use the markers to read out the values of all traces at a specific frequency? What you can do at the moment is this:
1. Create a marker for every trace that you want one one (e.g. for S11, S12, S21 and S22 with the default traces). Each individual marker will show up in its graph:
[image]
2. Link the created markers by selecting them in the marker dock and right-clicking:
[image]
3. The additional number in the marker dock identifies the linked markers (you can have multiple non-overlapping groups of linked markers):
[image]
4. Moving one marker also moves all others now.

It might be possible to create a shortcut for these steps (e.g. a button "create a marker for all traces and link all of them"). I am not sure where/how that should be implemented in the GUI. If someone has any suggestions about that please share them.

One small remark for the future: Please open separate issues for every improvement request. It might seem easier to simply write them into one issue (especially if they are small requests like these), but keeping track of them is a lot harder if different stuff is combined. Thanks :)

Chorbu, 2021-12-05:
Hi Jan,

Thank you so much for mods on your software ! i'm so happy to use your VNA !
I follow mods and discussions all days. Thanks for the community to share your fantastic project.
Specifications are under all other low cost VNA's i had used. (excuse me for my english...)

Here is my website : http://rflab.fr/

Best 73's Patrick

jankae, 2021-12-05:
Thanks, I am happy to hear that :)

I thought some more about the shortcut to create linked markers and I believe an additional button might do:
[image]
Clicking this button creates a marker for every trace and links them. It gets you the exact same result as the four steps I explained above but with a single click.

Chorbu, 2021-12-05:
Superb Jan,

Thank you !

Best 73's Patrick

closed 2021-12-05 by jankae