jankae/LibreVNA #75: Can't export trace as touchstone file

Germwalker, 2021-12-21:
Hello, 

I want to export S11 trace to a .s1p file.
But it's not working : any file is created. 
When i select a CSV export format, the file is created. 

How to reproduce ? Export -> Touchstone -> Select file content -> Select folder (desktop) -> Any file is created

I tried to execute the program as administrator, same thing. 

Configuration : 
Windows 10 - up to date 
LibreVNA GUI - v1.2.1

jankae, 2021-12-21:
Do you have a dot in the path anywhere? If so, it is probably this issue: https://groups.io/g/LibreVNA-support/topic/problem_with_saving_snp_files/87723114

12818d9 fixed that, could you please it try with the latest development version? (https://github.com/jankae/LibreVNA/actions/runs/1578629761)

jankae, 2022-01-07:
Closing due to inactivity - please let me know if this is still an issue

closed 2022-01-07 by jankae