jankae/LibreVNA #79: LibreVNA not recogniced under Lubuntu Linux & LibreVNA GUI 1.2.1

dk7xe, 2022-01-02:
LibreVna is not recognized under Lubuntu 20.04.3

dmesg shows 

```
[  632.474586] usb 1-1: new full-speed USB device number 5 using xhci_hcd
[  632.624029] usb 1-1: New USB device found, idVendor=0483, idProduct=4121, bcdDevice= 2.00
[  632.624034] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  632.624038] usb 1-1: Product: VNA
[  632.624041] usb 1-1: Manufacturer: STMicroelectronics
[  632.624043] usb 1-1: SerialNumber: 2072397D3350
[  882.058219] usb 1-1: USB disconnect, device number 5
[  883.729433] usb 1-1: new full-speed USB device number 6 using xhci_hcd
[  883.879603] usb 1-1: New USB device found, idVendor=0483, idProduct=4121, bcdDevice= 2.00
[  883.879610] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  883.879613] usb 1-1: Product: VNA
[  883.879616] usb 1-1: Manufacturer: STMicroelectronics
[  883.879618] usb 1-1: SerialNumber: 2072397D3350
```
but LibreVNA GUI shows no device after Update Device list

jankae, 2022-01-02:
At first glance, I can see two possible problems and solutions:
1. Your USB port does not provide enough current. The LibreVNA requires up to 1.5A which is violating the USB specification. If your port limits the current, the VNA will not fully start. Your dmesg output makes me think that this could be the problem, because it shows a diconnect followed by a connect. Please try another USB port, a powered hub or supply external 5V to the DC jack at the back of the VNA (power supply needs to support at least 1.5A).
2. A missing udev rule. This prevents the GUI from accessing the USB device and could be the issue of your "access denied" output. If you haven't already, please set up your udev rules accordingly. Copy [this](https://github.com/jankae/LibreVNA/blob/master/Software/PC_Application/51-vna.rules) file to /etc/udev/rules.d/

dk7xe, 2022-01-03:
Hi @jankae,
the missing udev rule was the problem.
Just tested on another Linux PC with the same result as reported initially. After applying the udev rule everything works as expected.
Would be nice to have that in the documentation for the Linux users. Thank you!

jankae, 2022-01-05:
I have changed the default behavior to display a warning message mentioning the udev rules if a potential device is detected but permissions are missing. Hopefully this is enough to point future users into the right direction.

closed 2022-01-05 by jankae