jankae/LibreVNA #80: [feature] Logarithmic frequency sweep/display

bbczeuz, 2022-01-04:
Hi,
I'd be interesting in an option to perform logarithmic frequency sweeps.
Option 1:
- Perform a linear frequency sweep (e.g. from 10MHz to 100MHz using 5 points: 10.000    32.500    55.000    77.500   100.000)
- Display x-axis in xy plot logarithmic (same as in a bode plot)
--> no changes in the embedded firmware needed

Option 2:
- Perform a logarithmic frequency sweep (e.g. from 10MHz to 100MHz using 5 points: 10.000    17.783    31.623    56.234   100.000)
- Display x-axis in xy plot logarithmic (same as in a bode plot)

Any chance, this is going to be implemented soon?
Is it possible to build the PC application without installing the whole Qt development environment?

jankae, 2022-01-05:
This has been on my list for a long time, your request made me start on this feature. The xy plot now has a log option for the X axis (only for the frequency domain, time/distance/power is always linear):
[image]
This is your "Option 1" and it only changes the way the data is displayed. For "Option 2", an additional checkbox is available in the sweep toolbar:
[image]
This performs a logarithmic frequency sweep. It is fully independent of the xy plot log/linear setting. There are still some issues with very high RBWs in log mode (calculation of frequencies takes more time for a logarithmic sweep. For RBW > 40kHz, the CPU is not fast enough anymore and misses points).

Example of a logarithmic sweep on a logarithmic plot:
[image]

This is a work in progress and I expect some more issues but feel free to take a look: https://github.com/jankae/LibreVNA/actions/runs/1658938707

> Is it possible to build the PC application without installing the whole Qt development environment?

Not as far as I know. If you want to build it yourself, I'd recommend to install the QtCreator IDE. If you just want an executable of the latest development version (without waiting for the next release), they are available here: https://github.com/jankae/LibreVNA/actions

jankae, 2022-01-07:
Further improvements for this feature: 48d00ec and 367451d

I am not aware of any bugs at the moment and consider it implemented. Please let me know if you discover any issues while using the logarithmic sweep

bbczeuz, 2022-01-08:
Hi Jan,
I'm very impressed by your work and speed!