jankae/LibreVNA #82: Proposal for adding continuos phase traces

oz7ou, 2022-01-08:
A continuos phase trace will "cleanup" the else "muddy" s21/s12 phase trace and aso be a tool for analyzing where phase measurement are problematic at specific frequences. More explained in the linked pdf file
[http://www.hamcom.dk/VNA2/Phase versus Continuos phase.pdf ](url)
kind regards
Kurt

jankae, 2022-01-09:
I have added an option to display unwrapped phase on the XY-plot (4d95959):
[image]
The wrapped and unwrapped phase of S11 in one plot as an example:
[image]

oz7ou, 2022-01-10:
Hi Jan

Thank you very much. Just installed and tested and it works great

Kind regards

Kurt

 

 <https://user-images.githubusercontent.com/16818866/148687220-ff2dc996-16b9-4167-b544-64dbd8924a2e.png> 
 <https://user-images.githubusercontent.com/16818866/148687183-1b25237b-aae8-42c5-8646-7211d91fcc73.png>

closed 2022-01-10 by jankae