jankae/LibreVNA #83: Load Calibration kit setting after the resent change

oz7ou, 2022-01-12:
I find it a bit misleading that for calibration setting of the load now is labelled as Z0 for the field to enter actual resistance of the load and the system impedance 50 ohm is hidden.
I think it would be better to have an entry for the system impedance Z0=50 ohm and then an entry field for resistance e.g. 51.2 ohm as well.
Kind regards
Kurt

jankae, 2022-01-14:
You are right, there should be a separate entry for the resistance. I have added that now: d5f2f70

What was previously called "Z0" in the dialog is now called "Resistance". The Z0 entry is still there, but now it is used to decribe the impedance of the transmission line in the load (from the SMA connector to the actual load resistor). Obviously it only has an effect if an offset delay is specified as well.

There is no need to modify old calibration files. If the file only contains the value for Z0, it will be used as the resistance instead and Z0 set to 50 Ohm (which results in the same behavior as with earlier versions of the GUI). However, new calibration files will contain both the Z0 and resistance setting.

oz7ou, 2022-01-14:
Thank you Jan, I will test right away

closed 2022-01-15 by jankae