jankae/LibreVNA #85: Data export improvement CSV

W0lpertinger, 2022-02-01:
Hi Jan,

all measured Data should be available to export in CSV.
Maybe an idea to let the user choose which data will be exportet with a check field of all availble kind of data.

BR

jankae, 2022-02-01:
Hi, can you explain what you mean by all available data? What data apart from the traces do you want to export?

You should be able to export all traces as CSV (but only traces which have the same frequency points in one file. If you have traces with different frequency points, you'll have to export multiple files)

W0lpertinger, 2022-02-02:
Hi, all available Data means in this case that all Data which can be choosen to be displayed in the table should be available in the CSV file. To have this information available for an export in CSV is common in projects like VNAJ, VNWA3 and also commercial like R&S. The CSV file should have rows like frequeny, magnetude, phase, impedance, imag, resistance, capacitance, inductance. Everey line will have the information from each measured point in a sweep. 500 Points = 500 lines in the CSV file.

W0lpertinger, 2022-02-02:
Example:

Frequency(Hz),Return   Loss(dB),Phase(deg),Rs,SWR,Xs,\|Z\|,Theta
14235000,17.77,5.10,64.8,1.30,1.5,64.8,1.3
14240000,18.18,5.10,64.0,1.28,1.4,64.0,1.3
14245000,17.71,5.10,64.9,1.30,1.5,64.9,1.3
...

jankae, 2022-02-02:
Okay, I got the idea, thanks. I'll have to think some more about it. As it is now, there could be multiple traces in one CSV file. With all these additional columns (maybe even with different columns per trace?) it could get a bit crowded and not so easy to import anymore.

jankae, 2022-04-08:
The CSV export now has an additional option, you can select the columns you want to export:
[image]
Depending on your selection it might not be possible to import the saved CSV file again (e.g. it is not possible to restore the trace data if you only exported the phase in the CSV)

closed 2022-04-20 by jankae