jankae/LibreVNA #87: Preferences File Crash

I-AM-EXFOLIATING, 2022-02-27:
While playing around with TDR on a coax cable, LibreVNA crashed.

When I restarted, the application worked fine, but trying to restore my window setup (adding a TDR trace) caused the app to crash again every time.

I removed my preferences file ($HOME/.config/LibreVNA/LibreVNA.conf) and tried again, this time everything works. The offending file is attached. I suspect this is probably something to do with loading a byte array directly without checking it.

[LibreVNA-GUI-bad.conf.txt](https://github.com/jankae/LibreVNA/files/8148700/LibreVNA-GUI-bad.conf.txt)

jankae, 2022-03-01:
Hi, thank you for the report. However, I couldn't get the app to crash with the config file you provided. Can you give me some more information to fix this please?
* Which version of the GUI are you using? (Version number + commit hash, you can find both in Help->About)
* After opening the app with your .conf file, what are the exact steps to get it to crash? E.g. what are the TDR settings you are changing? Screenshots might help here, otherwise a very detailed description might also work

I-AM-EXFOLIATING, 2022-03-01:
Version: 1.2.1-1c6a1ab6f

These are the exact steps I used to reproduce the issue:
1. Double-click S11 in Traces list
2. Click '+' symbol in Math operations
3. Select TDR in dialog then OK
4. Immediately after clicking OK the application crashes before showing any other dialog or activity of any kind.

However since generating a new config file I have not been able to reproduce the issue - even if I revert to the old config file.

jankae, 2022-04-08:
Any further crashes related to the preferences? If not, I will close this issue

I-AM-EXFOLIATING, 2022-04-09:
I haven't been able to repeat this behaviour.

closed 2022-04-09 by I-AM-EXFOLIATING