jankae/LibreVNA #89: Attempt to open VNA-GUI without the GUI creates QT platform error

MonsieurGamgee, 2022-03-02:
I'm attempting to piggyback on the GUI you've written to use the LibreVNA with an analysis script I've written in MATLAB. I'd like to just open up the VNA software in the background so that I can connect to the LibreVNA device through a TCP connection with MATLAB's instrument control toolbox functionality. However, when I attempt to use the command `.\mydirectory\LibreVNA-GUI_Windows-v1.2.1\release\LibreVNA-GUI --no-gui` in my windows command prompt, it creates an error:

> This application failed to start because no QT platform plugin could be initialized. Reinstalling the application may fix the problem. Available platform plugins are: windows.

I've looked up similar QT platform plugins problems via google, but so far I've only seen a recommendation that I make sure the qwindows.dll is in `C:\MyApp\platforms` (mine is in `.\release\platforms` so it should be fine). I'm curious if anyone else has had this issue, considering it's one of the examples in the programming guide, and if so, what they did to solve it. Anything is hugely appreciated. Thanks!

jankae, 2022-03-03:
Same problem here on windows (works fine with Ubuntu). I believe the issue was an ugly work-around I used to prevent the GUI from showing up (I manually set the "-platform offscreen" option for the main QT application if the --no-gui option was used). I have now changed the code not execute the show() or exec() functions for every window/dialog. This seems to have fixed it for me, please try with the latest version: https://github.com/jankae/LibreVNA/actions/runs/1927731579

MonsieurGamgee, 2022-03-03:
The new version works great for me now. Thanks so much!

closed 2022-03-03 by MonsieurGamgee