jankae/LibreVNA #90: Increasing the number of points limit

awesinha, 2022-03-04:
Hello,
I am doing some spectroscopy experiments and I would like to take more than the current limit of 4000 points.  Is that possible? It is too messy to have to stitch together multiple ranges and so it would be nice to have the ability to select the maximum number of points for a single sweep.
Thanks

jankae, 2022-03-05:
See answer in #91, this can't be done easily

jankae, 2022-03-07:
There might be a solution, see #91 again

closed 2022-03-09 by jankae

awesinha, 2022-03-09:
Hello,

Thank so much for taking care of the number of points limit.  Your solution works great and this is exactly what I needed.  We love the LibreVNA and it is wonderful piece of electronics.

It has a very wide frequency range and majority of people work in two broad frequency ranges ( < 100 MHz and > 100 MHz).  For > 100 MHz and GHz range, 100 Hz frequency resolution is more than adequate. However, for the < 10 MHz frequency region, this is somewhat limiting.

I am a physicist who works with ultrasound and use it for Swept Frequency Acoustic Interferometry measurements, a technique that I had developed a decade ago.  It produces a beautiful resonance spectrum with very sharp peaks. That is where 100 Hz becomes somewhat limiting.  Perhaps someday, you can some range specific frequency resolution.  What you have currently is perfectly fine for most applications and the LibreVNA is a beautiful piece of work.

Regards
Dipen

Closed #90<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fjankae%2FLibreVNA%2Fissues%2F90&data=04%7C01%7C%7Ca1cf40659956402d6ca508da01aaeb7b%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637824130848629718%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=XZatZnMRwUMvXvG9HVQPJVrWx7Y6F4%2B4y7aIvizPsA0%3D&reserved=0>.

-

jankae, 2022-03-13:
> 100 Hz frequency resolution is more than adequate. However, for the < 10 MHz frequency region, this is somewhat limiting.

Hi, could you explain the 100 Hz frequency resolution a bit more please? Internal frequency resolution is 1 Hz in every part of the software. However, the actual frequency resolution is limited by the PLL dividers (hardware limitation of the PLL chip). I am choosing the best available Frac/Modulus values for every frequency but there will be some deviation from the desired frequency

awesinha, 2022-03-13:
Hello,

I may be wrong about the 100 Hz frequency step as the minimum.  A programmer working on this told me that he thought the resolution he found to be 100 Hz and I did not double check.  A 1 Hz resolution is excellent.

Could you please check to see why the linear Y-scale does not work for S11 or S22 measurement?  In the GUI it comes checked as Linear although the display is in log-y.  This cannot be changed. I use your instrument for resonance spectroscopy and a log-display kind of ruins the display by deemphasizing the resonance peaks and showing the baseline noise more.  This is most likely a bug and I would appreciate if it can be fixed.  Others have bought the LibreVNA based on my recommendation and they also pointed out the same issue- No Linear scale display.

Thanks you very much.

jankae, 2022-03-13:
> A 1 Hz resolution is excellent.

Just a reminder that this is only the **software** limit. The actual hardware limit is different (higher) depending on your frequency range and exact frequency (hardware limit in the PLLs)

> Could you please check to see why the linear Y-scale does not work for S11 or S22 measurement?

The (default) setup of the graphs has a Y-scale that is "linear in dB" (which tends to be the standard for VNAs as far as I know). There are different options for the Y-scale available (right click on graph -> Axis setup). In any case, I am curious why you are mentioning S11 and S22 specifically. Is there any difference in terms of lin/log between them and S12/S21? This would indeed be a bug.

Also, this seems to have moved off-topic from the original issue. Please feel free to create a new issue for every different problem you encounter, that makes it easier for me not to miss anything.

awesinha, 2022-03-13:
Actually, all measurements show the same behavior - only Log and not linear.  It is not specific to S!! or S22.  Since most of the time I use these two modes, these were most important to me. When I said Linear - I did not mean ''Linear in dB', which is the same as log.  What I wanted was true Linear - plotting in  terms of Volts or whatever but NOT in dB.  I will create a new issue.  I appreciate that most engineers using a VNA requires display in dB.  Unfortunately, there are many other researchers, a display in Linear scale is more important where the shape of the resonance is more important that the dynamic range.

Thanks for being so responsive.  I really appreciate this.  You have a great product and this can be extremely useful in many R&D labs and universities.

jankae, 2022-03-13:
> It is not specific to S!! or S22.

Thanks, I got it now. I have added a "Magnitude (linear)" option to the graph:
[image]

The version with this addition can be found here: https://github.com/jankae/LibreVNA/actions/runs/1977639905

awesinha, 2022-10-11:
Hello,

There seems to be a bug in the GUI.  I hope you will be fix this easily.

  1.  The primary  y-axis log-scale is set permanently by default to log although it shows the linear box to be checked. It cannot be changed.  The phase is ok, as it is always in linear. Clicking on the Liner y-scale button does not seem to do anything.  Often one needs a linear y-scale instead of dB scale.  It is also hard to tell the x-log scale does anything or not.
There are other smaller features that can use some simplification and I will mention those later as I am mainly interested in a linear y-scale display.

Thank you.

Dipen

jankae, 2022-10-11:
> The primary  y-axis log-scale is set permanently by default to log although it shows the linear box to be checked.

Isn't that the same problem mentioned before? I consider the scale linear when the axis tick numbers are linear. In case of dB the ticks are linear, while the actual value is still logarithmic (because the conversion from raw values to dB introduces the log function). You can still select "Magnitude (linear)" as shown in the screenshot above, that should get you a true linear scale.

> Clicking on the Liner y-scale button does not seem to do anything.

Correct, the log option for y-axes is still not implemented, they are both fixed to linear at the moment. 

> It is also hard to tell the x-log scale does anything or not.

I can't confirm that. There is a very obvious difference when I switch between linear and log:
Linear:
[image]
Log:
[image]

This will be different if your span is small compared to your center frequency, but that is just a consequence of how the logarithm works mathematically.

> There are other smaller features that can use some simplification and I will mention those later

Yes, let me know about the other stuff but please open a new issue if it is not related to the original topic. This issue is about the number of sweep points and has been closed for a while. It is far easier for me to track open features if I don't have to scan old issues as well.