jankae/LibreVNA #91: Any way to capture more than 4501 points in a single sweep?

MonsieurGamgee, 2022-03-04:
My application for the LibreVNA sometimes requires more than the maximum allowed points of 4501 points over a wide frequency range. While I can manually stitch together smaller frequency range sweeps over that large range, it would be nice if I could just take a single shot over that whole frequency range with some 30000 points. Is the maximum allowed points in the GUI a limitation of the hardware or just a random cutoff that was put in? 

Thanks so much!

jankae, 2022-03-05:
The limit is the block RAM in the FPGA. For each point, the PLL settings and various other stuff must be stored. So I can't increase that easily, sorry.
One way would be to split up the sweep in the GUI and execute several sweeps with up to 4501 points each on the device. But this would slow down the sweep significantly, because the device needs to recalculate all the settings for each sweep.

MonsieurGamgee, 2022-03-05:
Ah I was hoping that wasn't the case. I'll have to make do. Sorry I didn't see the related issue. Thought I combed through all of them but apparently not.

jankae, 2022-03-07:
> But this would slow down the sweep significantly, because the device needs to recalculate all the settings for each sweep.

I was wrong, the slow-down penalty isn't that bad. Yes, the settings for each point need to be recalculated during the sweep (resulting in a noticable but reasonable short stop within the sweep), but I don't think it matters much as the sweep is slow when using that many points anyway.

Here are the improvements I implemented:
* When using 4501 or less points, nothing changes
* When using more points, the sweep is split into segments (with up to 4501 points per segment) by the GUI. Whenever a segment is completed, the settings for the next segment are transferred to the device. The LibreVNA device "thinks" it is doing a completely new sweep and the GUI stitches the data together.
* The new point limit (with segments) is 65535

You need to enable the use of segments in the preferences first, otherwise the old point limit applies:
[image]

I recommend to also update the firmware on the device as I have improved some PLL settling issues that showed up with the segmented sweep. The latest files (GUI + firmware) can be found here: https://github.com/jankae/LibreVNA/actions/runs/1948169516

MonsieurGamgee, 2022-03-08:
Wow, and you made it so it works with the --no-gui option as well! This is super helpful to me. Thanks!

A couple weird bugs with it, however. It seems like the new point limit is 32767, since anything over that number still just has the VNA spit out 32767 points. In addition, when I remove averaging (avgs = 1), the ':VNA:ACQuisition:AVGLEVel?' query returns 1 prior to reaching the 32767 points (actually prior to the 4501 point mark - I'm getting ~4300 (+/- 50) points on the S12 trace). The ':VNA:ACQuisition:FINished?" query similarly returns TRUE at the same point - prior to the 4501 mark. 

To clarify, the latter issue disappears when averaging 2 or more times.

jankae, 2022-03-08:
> In addition, when I remove averaging (avgs = 1), the ':VNA:ACQuisition:AVGLEVel?' query returns 1 prior to reaching the 32767 points (actually prior to the 4501 point mark - I'm getting ~4300 (+/- [...]

You are right, I used the wrong number of points for the averaging, this is now fixed.

> It seems like the new point limit is 32767, since anything over that number still just has the VNA spit out 32767 points.

I can't confirm this. I have tested it with the provided [script](https://github.com/jankae/LibreVNA/blob/master/Documentation/UserManual/SCPI_Examples/retrieve_trace_data.py), with two modifications:
* Set the IFBW to 10 kHz instead of 100 Hz
* Set the number of points to 40000 instead of 501

The script got all 40000 points after the sweep was finished. Can you share the script you are using?

MonsieurGamgee, 2022-03-08:
Found my error. I'd converted my npts to an int16 as opposed to keeping it as a double prior to feeding it into the string formatting tool. Must not have been playing nicely with the tool. I'm reading out 40000 points now. Thanks for the help!

closed 2022-03-09 by jankae