jankae/LibreVNA #94: S parameter renormalization

jankae, 2022-03-17:
I try the renormalization but I notice no change in the s params. 
My process is :
- Calibration SOLT in 50 ohm
- DUT 75 ohm connected
- I got the s params in 50 ohm reference, but if I pu 75 ohms the s params are the sames after a new acquisition
I do something rong ?
Thx

_Originally posted by @HadesDT in https://github.com/jankae/LibreVNA/issues/2#issuecomment-1070841559_

jankae, 2022-03-17:
My bad, I wasn't thinking far enough. You are not doing something wrong. As it is now, the reference impedance only influences the conversion from S parameters to complex impedances (e.g. the center of the smith chart would read as 75 ohms in your case). Nothing in the S parameters is changed, I assumed you would calibrate with a 75 ohm load as well (but even that wouldn't work properly). Looks like this feature needs more work - it is a bit difficult find all the implications this has in the different parts of the software.

HadesDT, 2022-03-17:
Ok, thx for your support :)

jankae, 2022-03-17:
Next attempt: I have moved the renormalization from the preferences to the de-embedding options (De-embedding->Setup...->add "Impedance Renormalization"). It now affects the S parameters and a whole lot of other things. Essentially each trace is now aware of its reference impedance (you can view it as a read-only field in the trace properties). This has some implications, e.g. you can not mix traces with different reference impedances on the same smithchart or export them in the same touchstone file.

HadesDT, 2022-03-18:
Hello,

I just tried the new version and looks to work perfectly for the renormalization !

Thx for your work !

jankae, 2022-03-18:
okay, thanks for the confirmation

closed 2022-03-18 by jankae