jankae/LibreVNA #95: LibreVNA Raspberry Pi 4 Model B

jresch, 2022-03-27:
Hello,

I was curious if anyone has used the LibreVNA with a raspberry pi...I'm having some issues with the software. 
Any help would be greatly appreciated :)

jankae, 2022-03-27:
Hi, what kind of issues are you having?
I have tried a couple of times, but so far I have been unable to compile it for a raspberry (unable to set up the cross-compile toolchain correctly

jresch, 2022-03-27:
Appreciate the fast response. 
So in that case, I guess you'd have to use python scripts running on the raspberry pi to communicate with the hardware?

jankae, 2022-03-27:
The GUI must run on the same machine that is connected to the LibreVNA via USB. I suppose you could connect the LibreVNA to a PC and control that with the Raspberry but that might not be what you are looking for. I am running a native compilation on a Pi 3 at the moment (I do not have an available Pi 4). So far, there are no errors, it has been running for about an hour. I expect results tomorrow.

jresch, 2022-03-28:
That's awesome to hear!
Were you planning on posting?

jankae, 2022-03-28:
The compilation finished after about 3 hours. No errors were reported and the GUI starts on my Pi. The Pi probably doesn't have the required performance for more demanding calculations (e.g. a TDR with a high IF bandwidth), but the basic sweep works great. This is the compiled executable:
[LibreVNA-GUI_RPi3.zip](https://github.com/jankae/LibreVNA/files/8361201/LibreVNA-GUI_RPi3.zip)
I don't know if it will run on a RPi4 as well, but it might be worth a try.

jresch, 2022-03-28:
I will check it out and let you know :)

jresch, 2022-04-04:
Sorry it took so long to reply ... had software issues with the Raspberry Pi, but your LibreVNA-GUI_RPi3.zip seems to work! About to do some test measurements now!! :)

jresch, 2022-04-04:
Woot!! Seems to be working! Thank you so much!
 
 The thing I'm trying to do now is write a script so I can take multiple measurements with my DUT and save all the scattering parameters (phases and magnitudes) in an .s2p file. I'm looking through the functions in your guide and it seems like it would be possible. Am I correct in this assumption?

jankae, 2022-04-04:
Happy to hear that the GUI is working on the Raspberry :)

Yes, you can take measurements via a script. Here is an example on how to set up the measurment: https://github.com/jankae/LibreVNA/blob/master/Documentation/UserManual/SCPI_Examples/retrieve_trace_data.py

You can either extract the trace data for each paramater (S11,S12,S21 and S22) as shown in the example or you can request the touchstone file itself (see the VNA:TRACe:TOUCHSTONE command in the programming guide). I haven't used the touchstone command since I implemented and tested it, but maybe I managed to not break anything since then ;)

jresch, 2022-04-05:
Another question @jankae  - I saw online when I bought the VNA that it can do measurements for up to 10,000 points, but I can't seem to set it for more than 4,501 points. Is this its max?

Thanks :)

jankae, 2022-04-05:
It is 10.000 points **per second**. The hardware is limited to 4501 points per sweep. Recently I added the option to use more points per sweep in the software, see #91

closed 2022-04-08 by jankae

EmilJankes, 2022-06-07:
> Sorry it took so long to reply ... had software issues with the Raspberry Pi, but your LibreVNA-GUI_RPi3.zip seems to work! About to do some test measurements now!! :)

How did you execute the file? I can't get it to work with ubuntu on RasPi 400. It's not responding when I'm trying to open the executable.

jresch, 2022-08-01:
> 

@jankae did you do the compilation for buster vs bullseye? I have tried to run it for buster and it doesn't work.

jankae, 2022-08-01:
I don't really know anymore but would have guessed buster. Can you elaborate on "it doesn't work" a bit more?