jankae/LibreVNA #96: Add the possibility of entering EMI/EMC standard limit for Spektrum Analyser

HadesDT, 2022-03-29:
Hi,

One proposal of improvment on the Spektrum side.

- Add the possibility of entering EMI/EMC standard limit values to set, for example, a CE102 limit and see if the setup passes the EMI/EMC standard. As on the Spektrum R&S, see print screen

[image]
[image]

Cheers,

sp9bsl, 2022-03-29:
Hi,
I would adwice some caution using LibreVNA as spectrum analyzer, especially for EMI/EMC measurements. Jan mentioned the way SA mode works in LibreVNA, it just remove some of the aliases during measurements. Another thing, for EMC measurements you need specyfic time for averaging and of course CISPR filter shape. LibreVNA is the best thing for network measurements, not for SA application. I use daily R&S stuff like ESR, ESL with GTEM/TEM and LISN. Our great little thing is not very usable for such measurements and may do some confusion reading the results...

HadesDT, 2022-03-29:
Hello,
I'm in phase with your comment.
It's just a proposal to add limit manually to pre-test some setup. Of course for the real test in EMI/EMC room we use real labo equipment.

jankae, 2022-04-22:
Hi,
although I agree that the LibreVNA probably is not accurate enough or does not have the correct filter for good EMI/EMC measurements, I think limit checking is a worthwhile feature and implemented it:
[image]
Setting it up works like this:

- Configure constant lines in a XY-plot (right-click->setup)
- Set the constant line points according to your limit and select whether this is a max or min threshold (high limit or low limit)
- You can choose the pass/fail indicator behavior under window->preferences->graphs->limit checking

Here is also the setup file of the example from my screenshot:
[limit.zip](https://github.com/jankae/LibreVNA/files/8540683/limit.zip)

The pass/fail result is also exposed via the SCPI API, however the actual configuration of the limits must be done in the GUI (not possible via SCPI).

HadesDT, 2022-04-25:
Hello,

I try this version and works perfectly ! Thx for your job !

jankae, 2022-04-25:
Thank you for confirming that it works like this for you :)

closed 2022-04-25 by jankae