jankae/LibreVNA #97: TDR seems to calculate wrongly

dotmjsc, 2022-04-03:
Look at the attached picture.
Open is displayed as short, short as open and 75R as 30R. Seems to be a minor calculation error (or I do not get how to configure this?)
Thank you.
M.
[image]

jankae, 2022-04-03:
Hi, it is possible to configure the TDR wrong but I can't tell just by the screenshots. Can you please provide the setup file (File->Save Setup) and also a touchstone file with your S parameter data? That way I should be able to replicate your calculations for the TDR.
TDR seems to work here after a quick test (30cm coax terminated with a short):
[image]

dotmjsc, 2022-04-03:
Wow, fast answer! Thanks.
Here is the desired data for 30cm with sma short.
[TDR_bug.zip](https://github.com/jankae/LibreVNA/files/8405094/TDR_bug.zip)

jankae, 2022-04-03:
Your setup looks correct. However, your S11 data is approximately "open" (reflection coefficient +1) at its lowest frequency (1 MHz). This is the reason why your TDR shows it as an open as well.

From your data I can't tell why your S11 is measured that way. This could have happened if the short and open standards were mistakenly swapped during the calibration. Can you repeat your calibration and measurement please? In any case, this points to an error/bug in the S11 acquisition. Any subsequent TDR calculations seem correct to me.

dotmjsc, 2022-04-03:
I never did a calibration after receiving the unit. Will repeat it right now and try again!

dotmjsc, 2022-04-03:
Ok, That was it, thank you! After doing the calibration the trace looks exactly the same as yours.

closed 2022-04-03 by dotmjsc