jankae/LibreVNA #98: Feature Request: Font support

kelemvor4, 2022-04-10:
Please consider adding support for OS installed fonts or simply font scaling.  Currently librevna-gui uses a very tiny font that does not scale with OS scaling settings on windows.  It makes the software very hard to use.

jankae, 2022-04-11:
Thanks for the suggestion. Does this apply to all fonts used in the GUI? I am certain that text on the graphs will not scale (as I am drawing it myself with a fixed size) but I would have expected that Qt honors font scaling in windows (e.g. the menu, labels on buttons,...).
Getting the font scaling right on the graphs might be tricky and could potentially result in overlapping texts unless done carefully. I will look into it but this issue probably will not get the highest priority at the moment, sorry

kelemvor4, 2022-04-14:
Yes,  just an option to choose the point size or scaling would do the trick.  Changing the actual font might be nice but simply changing scaling would do the trick.   

Totally get that it's not a high priority, just thought it should be on your radar for a rainy day.

jankae, 2022-04-20:
And here it is: (in the preferences)
[image]
You can select the font sizes of different stuff on the graphs (size is the font height in pixels). This does not affect the font size in the rest of the application (e.g. text on buttons) as this is handled by Qt itself.

jankae, 2022-05-09:
I'll go ahead and close this. If I missed some font please let me know

closed 2022-05-09 by jankae