jankae/LibreVNA #99: LibreVNA-GUI v1.2.1 Markers lost after changing graph settings

kelemvor4, 2022-04-10:
If you create a marker and have that marker pointed at a frequency that is then hidden by a change in sweep range, the marker becomes inaccessible.  There's no way to remove the inaccessible marker other than to discover where it was and change the graph so it re-appears.  Then it can be moved or deleted.  You could either automatically move the markers to be within range or allow right click and delete on the marker data displayed to the right of the graph or you could simply remove all markers upon changes which would cause it to be out of scope.

kelemvor4, 2022-04-10:
Example:   

1. Create a graph measuring S11 VSWR from 100Mhz to 2Ghz.   
2. Set a marker at 1.5ghz.
3. Change the sweep stop to 1.0ghz.  

You now have an inaccessible marker.

jankae, 2022-04-11:
After step 3, my GUI looks like this:
[image]
You can still access the marker in the marker dock (bottom right) and change its position or delete it.

kelemvor4, 2022-04-14:
Thanks, I didn't even see the box down there.

closed 2022-04-14 by kelemvor4