jankae/LibreVNA #101: SCPI Touchstone issues and remote Cal kit load

MEMS-lab, 2022-05-03:
I am having issues reading data in the Touchstone format. When I run:
data = vna.query(":VNA:TRACE:TOUCHSTONE? S11 S12 S21 S22")
I get: data = '# GHZ S RI R 50.000000000000'
and nothing else. Reading individual data works fine though:
data = vna.query(":VNA:TRACE:DATA? S11")

Any ideas? I am running the latest Windows GUI v1.3.0 and firmware rev-B-v1.3.0.

A feature I am missing is to set the cal kit for the calibration. Is there a command that is not provided in the programming manual? 
While I am at it, would it be possble to add a ~160° phase shift to the ports? This way on power up the open S11 and S22 would look more like opens and shorts.

As critical as my comments might seem, I am very impressed with the LibreVNA! Excellent work and great documentation, except being stuck on the two issues. Many thanks for comments.

jankae, 2022-05-04:
> I get: data = '# GHZ S RI R 50.000000000000'
and nothing else.

It's not a bug it's a feature. Well, not really. Someone asked me to include that touchstone command, so I did. However, the touchstone file includes newlines. And the example SCPI client implementation uses newlines to determine the end of a response. So you will only get the first line. You will have to write your own SCPI client or modify the existing example to receive the complete touchstone file. Another way could be to grab the data of all traces one by one (with :VNA:TRACE:DATA?) and assemble the touchstone file yourself.

> A feature I am missing is to set the cal kit for the calibration. Is there a command that is not provided in the programming manual?

There isn't a specific command for it, but saving/loading a calibration implicitly saves/loads the cal kit. Whenever you save a calibration, a .calkit file with the same name is also created. When you open that calibration (through the GUI or SCPI), that .calkit file will be used.

> While I am at it, would it be possble to add a ~160° phase shift to the ports?

I assume you want the smith chart to look nicer when the device is running without any calibration? I don't really see the point, sorry. As soon as you have a correct calibration, this problem doesn't exist anymore. And without a calibration, you can't trust the measurements anyway.
You could probably achieve your desired behavior by assigning a default calibration (Device->Default Calibration). This links a specific calibration file to the serial number of your LibreVNA. Whenever you connect that LibreVNA, the GUI will automatically load the specified calibration and S11/S22 will show the correct data right from the start.

> As critical as my comments might seem, I am very impressed with the LibreVNA! Excellent work and great documentation, except being stuck on the two issues.

Thank you very much :) I am trying my best. Please let me know if you find any further issues or have improvement ideas.

MEMS-lab, 2022-05-05:
Hi jankae,
Many thanks for the immediate response. 
> touchstone: your comment was very helpful. I need to find some time, but then adding a function in the libreVNA.py that does not use the readline() command that ends the read as you stated on \n will fix it. The workaround reading the four Sij traces works fine.
>loading the cal kit through referring to a saved calibration is awesome! I was not aware of this relation. So solves my problem! Many thanks.
>Optics on an open in the Smith Chart: yes, it is pure optics and I get your purist point :  ) It is what it is. I will look into the default calibration. Thanks for pointing that out. Especially when training/teaching how to use a VNA and when people are not too familiar with Smith charts, especially when connecting cables, cal standards etc., I think it is critical to watch the Smith chart as you connect devices and I find it a bit confusing on sweeps starting near DC to start in the wrong region. Having seen broken VNAs power up, when I connected the Libre VNA for the first time, I thought I had wasted money on this ;  ) Now, I am happy of course owning one, but I am sure I am not the first and last to judge rather quickly. Many thanks again for the super fast and very helpful responses!

closed 2022-05-05 by MEMS-lab

jankae, 2022-05-07:
> [...] and I find it a bit confusing on sweeps starting near DC to start in the wrong region. Having seen broken VNAs power up, when I connected the Libre VNA for the first time, I thought I had wasted money on this

Okay, I understand the issue a bit better now. It would be fairly trivial to add the phase shift to the ports. The only problem I have with it is that it would mess any calibrations done before I do the phase shift (e.g. a user does a firmware update, gets the phase shift and opens an old calibration -> completely wrong values). Not sure yet how I will proceed, maybe add a flag to the calibration with the VNA firmware so I can show warnings when using an old calibration with a new firmware in the future?