jankae/LibreVNA #104: Signal generator stepping frequency by mouse scroll wheel

oz7ou, 2022-05-19:
I would be a nice feature it the frequency could be manually stepped by scrolling the mouse whell. The step size e.g. by selecting the mouse position on the frequency entry field or by a selection in a drop box.
Kind regards
Kurt

jankae, 2022-05-19:
Hi Kurt,

> The step size e.g. by selecting the mouse position on the frequency entry field

It already works like that: click in the frequency field and position the cursor behind the digit you want to change. Then use the mouse wheel to change that digit. This works for every entry field that uses the internal SIUnitEdit class (which is almost every entry field throughout the GUI)

oz7ou, 2022-05-19:
Hi Jan

Thank you very much for this excellent information. I did try so a couple of times and now I can trace why it did not worked so.

Kind regards 

Kurt

closed 2022-06-06 by jankae