jankae/LibreVNA #108: Using SCPI to save data output

jresch, 2022-05-30:
Hi @jankae, it is me again!

Do you recall when I asked you about writing a script so I can take multiple measurements with my DUT and save all the scattering parameters (phases and magnitudes) in an .s2p file? 

Well, I tried to do this using the `data = vna.query(":VNA:TRACE:TOUCHSTONE? S11 S12 S21 S22")` command and save the data to a file, but it only printed out  `# GHz S RI R 50. ` Perhaps I'm doing something wrong?

I have attached the file that gave the best results, but even these aren't ideal, as it would be best to have the s2p file or the scattering parameters in columns (with the frequency being the first followed by the magnitude and phases for each scattering parameter).

[retrieve_trace_data_extension_code.zip](https://github.com/jankae/LibreVNA/files/8799807/retrieve_trace_data_extension_code.zip)

Any help would be appreciated!

jankae, 2022-05-30:
Have a look at #101, that should explain what is going on. It is a known issue and there are work-arounds possible but no super easy solution I am afraid.

jresch, 2022-05-30:
Okay, thanks very much!

closed 2022-06-06 by jankae

jankae, 2023-03-02:
> I found out that each execution of the ":VNA:TRACE:TOUCHSTONE?" query returns a single line of the touchstone file.

No, it returns the whole touchstone file. But the SCPI receiver (the example python implementation) assumes that the answer is complete after it detects the first newline. It only returns that line and leaves the rest of the file in the receive buffer. On the next command, the already received next line of the touchstone file is returned instead (and the actual answer to the command added all the way at the end of  the receive buffer).

I provided a more detailed explanation in #101. I suggest you adjust the SCPI receiver to read multiple lines after the touchstone command (probably with a timeout to detect the end of file?).

abgut, 2023-03-03:
Thanks for clarification. Now it makes much more sense.