jankae/LibreVNA #110: Signal ID fails at certain frequencies with small span hiding real signals

erikkaashoek, 2022-06-07:
This was tested with the latest FW and PC application and all settings reset to default.
In SA mode, at certain frequencies a signal disappears when reducing the span.
Example frequencies:
1570000000
1700000000
1830000000
1960000000
2090000000
2220000000
2350000000
2480000000
2610000000
2870000000
3000000000
3130000000
3260000000
3390000000
3520000000
3650000000
3910000000
4170000000

The attached 3 images show the stepwise disappearing of the signal with decreasing span
10MHz span
[image]

1MHz span
[image]

100kHz span
[image]

jankae, 2022-06-07:
All settings reset to default means that you are using an RBW of 10 kHz? Could you share the setup file please? Probably easier than confirming every setting by text.

erikkaashoek, 2022-06-07:
Settings
[settings.zip](https://github.com/jankae/LibreVNA/files/8852096/settings.zip)

RBW was set to 1kHz

jankae, 2022-06-07:
I was unable to replicate the issue (using your setup file, enabling signal ID and only changing the span):
10 MHz span:
[image]
1 MHz span:
[image]
100 kHz span:
[image]

What are you using as a signal source?

erikkaashoek, 2022-06-07:
You are not testing one of the frequencies in my list. It only happens at specific frequencies.
Try 3GHz, it only happens on 3GHz +/- some kHz

jankae, 2022-06-07:
Ah, my bad. Yes, at 3 GHz I also see the problem. Probably another issue of the PLL unable to reach the exact frequency, I'll try to fix it soon.

erikkaashoek, 2022-06-07:
The problem appears every 13MHz

jankae, 2022-06-07:
Yes, that fits the suspected PLL divider issue (PLL reference frequency is 104 MHz, 8 * 13 MHz).
I was actually already compensating for the PLL divider accuracy by shifting the 2.LO. But I forgot to adjust the direction of the shift during the signal ID steps (1.LO sits on either side of the signal frequency depending on the signal ID step). The problem is hidden during larger spans because there are actually multiple frequency points captured per displayed point. If at least on point doesn't have the problem, the signal will show up (assuming a positive peak detector).

Solution: check if 1.LO is below or above of the frequency of interest and shift 2.LO accordingly. During my tests, all signals showed up now. Also fixed some smaller other issues while being in there.

erikkaashoek, 2022-06-07:
I've tested the new FW and all seems to be OK, will you close the issue?

jankae, 2022-06-07:
Yes, thank you for confirming

closed 2022-06-07 by jankae