jankae/LibreVNA #111: Load setup adds new toolbars

sp9bsl, 2022-06-08:
Hi Jan,
using LibreVNA is a great pleasure for me, thank you for your hard work. This is my basic instrument for microwave experiments however I found a kind of bug in software: when I load setup saved previously, it adds new toolbars which are moved to right (out of the screen).
Thus after load I have second set of  Sweep type, Level and Calibration. I can't close the set created at start and I can't use it to set parameters. For each time I load new setup I must slide the newest toolbars to the left.

Slawek

jankae, 2022-06-08:
Hi Slawek, seems like a duplicate of #103.
If so, this has been fixed and it is probably time for another release so no one else runs into this bug ;)

sp9bsl, 2022-06-08:
Ahh ok, I missed notification about this bug :) Thank you, I'll wait for new release.

closed 2022-06-08 by sp9bsl