jankae/LibreVNA #112: Frequency returned in :DATA command over SCPI is rounded as float

Ivesvdf, 2022-06-12:
When using LibreVNA-GUI v1.3.0, open the default config (VNA, Freq sweep) and set center to 4G and span to 500Hz. 

Over SCPI, send the command :VNA:TRAC:DATA? S21
String returned  (cut off)
`:VNA:TRAC:DATA? S21
[4e+09,0.00109066,-0.00826269],[4e+09,0.00106264,-0.00826557],[4e+09,0.00106017,-0.00824748],[4e+09,0.00103726,-0.00827891],[4e+09,0.00105232,-0.00827352],[4e+09,0.00105631,-0.00827812],[4e+09,0.00105561,-0.00825316],[4e+09,0.00107841,-0.00825511],[4e+09,0.00104889,-0.00825346],[4e+09,0.00107413,-0.00826114],[4e+09,0.00105683,-0.00827255],[4e+09,0.00108391,-0.00826884],[4e+09,0.00106727,-0.00824737],[4e+09,0.00105697,-0.00824445],[4e+09,0.00107071,-0.00827314],[4e+09,0.00107066,-0.00826179],[4e+09,0.00105522,-0.00826983],[4e+09,0.00105325,-0.00826495],[4e+09,0.00108215,-0.00828562],[4e+09,0.00105993,-0.00826886],[4e+09,0.00106004,-0.00827988],[4e+09,0.00106923,-0.00826772],[4e+09,0.00106009,-0.00829076],[4e+09,0.0010573,-0.00828201],[4e+09,0.0010425,-0.00828302],[4e+09,0.00105672,-0.00827506],[4e+09,0.00106062,-0.00827384],[4e+09,0.0010584,-0.00827589],[4e+09,0.00102987,-0.00828059],[4e+09,0.00103959,-0.00824771],[4e+09,0.00104123,-0.00827128],[4e+09,0.00107619,-0.00828725],[4e+09,0.00105861,-0.0082517],[4e+09,0.00106211,-0.00826979],[4e+09,0.00108088,-0.00826677],[4e+09,0.00105199,-0.0082439],[4e+09,0.00106942,-0.00827846],[4e+09,0.00107022,-0.00827809],[4e+09,0.00105323,-0.0082708],[4e+09,0.00106979,-0.0082566],[4e+09,0.00108446,-0.00825782],[4e+09,0.00110101,-0.00827666],[4e+09,0.00105985,-0.00825567],[4e+09,0.00108531,-0.00826824],[4e+09,0.00105903,-0.00827829],[4e+09,0.00107591,-0.00828398],[4e+09,0.00108157,-0.00828448],[4e+09,0.00108776,-0.00830153],`

It can be noted that the first field in each triplet is identical for all datapoints, probably due to it being stored as a floating point number and rounding. 

In the UI however, it looks like the device actually does perform a sweep:
[image]

Expected behavior is that the frequency component returned will be different for all measurement points, as we are doing a 500Hz sweep with 500 points, so approximately 1Hz per point.

jankae, 2022-06-12:
Thanks for the report. This should be fixed now by d701ed6. The frequency was actually stored as a double value internally, but the output function (QString::number) didn't have the required format setting for enough precision.

Ivesvdf, 2022-06-12:
Excellent, thank you! Closing as complete.

closed 2022-06-12 by Ivesvdf