jankae/LibreVNA #113: SCPI Command and query VNA:ACQ:SINGLE always seems to return ERROR

Ivesvdf, 2022-06-12:
Steps to reproduce: open LibreVNA GUI v1.3.0, with the LibreVNA on the latest published FW image. Open default config, meaning VNA with sweep type set to frequency. 

Over SCPI, execute the following commands
[image]

Every possible variation of the single command fails with "error". I also can't find any reference to single mode in the UI, so I'm unsure how to test it through the GUI. 

Expected behavior: 
See reference manual
[image]

jankae, 2022-06-12:
Yes, that is the expected behavior with that version. The "single"-command was added by d84d3e80, which happened after the 1.3.0 release. Until I create the next release, you'll need to use the latest development version (download compiled versions of each commit here: https://github.com/jankae/LibreVNA/actions)

Ivesvdf, 2022-06-12:
Ok great, thank you. Closing.

closed 2022-06-12 by Ivesvdf