jankae/LibreVNA #118: When 0Hz VNA span is selected, only single datapoint is returned but device seems to perform entire sweep

Ivesvdf, 2022-06-19:
When specifying a 0Hz span (pretty much running the VNA in CW mode, which is useful for our use case where the DUT changes continually), only one point is returned over SCPI:

[image]
[image]

At the same time, the sweep seems to still take just as long as when multiple points are returned. This leads me to believe the sweep of 500 points is still performed, somehow aggregated.

I would expect that when 500 points are specified, that 500 points are returned, Of course all points will have the same frequency. 

If this is intended behavior, an alternative would be a CW sweep type I suppose, though that's more work on the implementation side I'm sure. CW operation of a VNA is supported by competitors like CopperMountain. 

A possible work-around could maybe be to specify only a single point and do repeated scans, though I'm guessing there will be significant overhead in the processing increased dead time. I'm aware there is a CW output in the software, but according to the manual there are no measurements in this mode.

jankae, 2022-06-19:
Zero Span is not supported at the moment, sorry. The sweep is actually executed with the requested number of points but each point will override the last in the GUI (as they are at the same frequency). The whole concept of traces is based around distinct frequencies, from adding new points, to handling of markers, displaying of data on the graphs
Unfortunately, it is not so easy to change that. There are several changes that are required to do so, e.g.:

- Handle incoming trace data based on the point number and not based on the measurement frequency
- Add a new time-domain for zero span sweeps (you need time as the x-axis for the graphs, otherwise you will just see a dot/line). This probably needs to be different from the already implemented time-domain which can be used for TDR measurements (frequency-domain transformed to time-domain)
- Ideally, use CW sweep for zero span mode (only possible if the source signal only needs to be at one port). This is probably the easiest task, the embedded firmware might even already support everything for that (I need to check again).

All in all, this is a significant amount of work, but zero span sweeps have been on my mind for some time. As there is not much to do on the hardware side at the moment (lack of parts), I might take it on soonish.

Ivesvdf, 2022-06-19:
Yeah I figured it was something like that on the back-end of the software.

I checked in S2VNA for CopperMountain, and indeed in LogMag view the horizontal scale changes to time when 0 span is selected. In Smith chart it just displays as normal in a sweep. 

Thanks for the update.

jankae, 2022-06-22:
It is a work in progress, but you can try this version for zero span: https://github.com/jankae/LibreVNA/actions/runs/2545350113

Probably a couple of bugs still in there, though...

Ivesvdf, 2022-06-23:
Thank you, I'll try this weekend.

Ivesvdf, 2022-06-26:
As promised, I spent some time testing the 0-span functionality. I tested with the GUI from the run you provided. It also recommended I update the FW as protocol versions were not matching, so I did update the FW image using the same build run. 

I also tried with the latest Windows GUI build on master but got an error on startup. 

In the UI CW mode at first seemed to be working fine. One small remark, over SCPI:

```
:VNA:SWEEP FREQUENCY

:VNA:FREQ:CENT 4700000000

:VNA:FREQ:SPAN 0

:VNA:TRAC:DATA? S21
[0,-0.000556021,-0.00962514],[0,-0.000532221,-0.00965229],[0,-0.000542607,-0.00963015],[0,-0.000549363,-0.00961872],[0,-0.000595413,-0.00964943],[0,-0.000537909,-0.00964565],[0,-0.000534994,-0.00964844],[0,-0.000582507,-0.00961861],[0,-0.000575209,-0.00961072],[0,-0.000563168,-0.0096132],[0,-0.000545197,-0.00964848],[0,-0.000570176,-0.00961296],[0,-0.000562308,-0.00961313],[0,-0.000517064,-0.009664],[0,-0.000563343,-0.00963249],[0,-0.000531725,-0.00962279],[0,-0.000540677,-0.00963163],[0,-0.000549144,-0.00963023],[0,-0.000553059,-0.00964919],[0,-0.000506389,-0.00965108],[0,-0.000517297,-0.00966532],[0,-0.000551651,-0.00966734],[0,-0.000549228,-0.00963034],[0,-0.000504819,-0.00967602],[0,-0.000530819,-0.00964454],[0,-0.000558798,-0.00963518],[0,-0.000538476,-0.00961977],[0,-0.000584563,-0.00964536],[0,-0.000557518,-0.00961643],[0,-0.000533861,-0.00963386],[0,-0.000555865,-0.0096208],[0,-0.000544477,-0.0096213],[0,-0.000558429,-0.00964338],[0,-0.000566445,-0.0096403],[0,-0.000526603,-0.0096413],[0,-0.000532269,-0.00964185],[0,-0.000561499,-0.00962465],[0,-0.000524752,-0.0096159],[0,-0.000510037,-0.00963262],[0,-0.000532287,-0.0096202],[0,-0.000507567,-0.00961271],[0,-0.000513157,-0.00960865],[0,-0.000572753,-0.00962163],[0,-0.000556,-0.00961779],[0,-0.000505936,-0.00959877],[0,-0.000579718,-0.00963571],[0,-0.000508385,-0.00963999],[0,-0.000514807,-0.00964913],[0,-0.000567773,-0.00961856],[0,-0.000507301,-0.00965342],[0,-0.000503716,-0.00966139],[0,-0.000545424,-0.00964586],[0,-0.000543069,-0.00961943],[0,-0.000534761,-0.00964799],[0,-0.000524118,-0.00962966],[0,-0.000522243,-0.00961333],[0,-0.00052285,-0.00965152],[0,-0.000530103,-0.00965767],[0,-0.000518404,-0.00964755],[0,-0.000544777,-0.009651],[0,-0.000550296,-0.00965148],[0,-0.000545964,-0.00967036],[0,-0.000526331,-0.009622],[0,-0.000580327,-0.00964244],[0,-0.000535775,-0.0096201],[0,-0.000536194,-0.00960222],[0,-0.000524596,-0.00962352],[0,-0.000535972,-0.00962908],[0,-0.000519685,-0.00964662],[0,-0.000530586,-0.00962427],[0,-0.00057994,-0.00959708],[0,-0.0005065,-0.00963562],[0,-0.000514111,-0.00963439],[0,-0.000539494,-0.00962498],[0,-0.000542456,-0.00964294],[0,-0.000522529,-0.0096701],[0,-0.000558787,-0.00963032],[0,-0.000571661,-0.00962386],[0,-0.000534176,-0.00964087],[0,-0.000509241,-0.0096204],[0,-0.000561218,-0.00961431],[0,-0.000501713,-0.00962115],[0,-0.000553447,-0.0096372],[0,-0.000532755,-0.00965677],[0,-0.000536499,-0.0096316],[0,-0.000538337,-0.00966645],[0,-0.000553288,-0.00962598],[0,-0.000566403,-0.00966216],[0,-0.000541965,-0.00965254],[0,-0.000553547,-0.00964993],[0,-0.000568945,-0.00962172],[0,-0.000524674,-0.00964057],[0,-0.000549944,-0.00961314],[0,-0.000564215,-0.00962493],[0,-0.000538281,-0.0096179],[0,-0.000516006,-0.00966011],[0,-0.00054986,-0.00963634],[0,-0.000532994,-0.00962991],[0,-0.00053513,-0.00966273],[0,-0.000517455,-0.00966431],[0,-0.000556685,-0.00962027],[0,-0.000508483,-0.00961158],[0,-0.000544771,-0.009634],[0,-0.000513279,-0.00965659],[0,-0.000513354,-0.00959499],[0,-0.000521493,-0.00963406],[0,-0.000534838,-0.00962968],[0,-0.000539396,-0.00965327],[0,-0.000586663,-0.00963576],[0,-0.000508283,-0.00966221],[0,-0.000533616,-0.00963791],[0,-0.000552634,-0.00964996],[0,-0.000569613,-0.00964108],[0,-0.000551857,-0.00966704],[0,-0.000569162,-0.00965085],[0,-0.000555713,-0.00963991],[0,-0.000537385,-0.00961158],[0,-0.000508421,-0.0096552],[0,-0.000564524,-0.00964649],[0,-0.000588476,-0.00964136],[0,-0.000546692,-0.00965938],[0,-0.000558747,-0.00959982],[0,-0.000574949,-0.00964101],[0,-0.000559305,-0.00958452],[0,-0.000566268,-0.00962966],[0,-0.000552449,-0.00962402],[0,-0.000542809,-0.00964104],[0,-0.000570166,-0.00964424],[0,-0.000502526,-0.00961548],[0,-0.000529296,-0.00961683],[0,-0.000578829,-0.00962724],[0,-0.000535236,-0.00961109],[0,-0.000526847,-0.00962547],[0,-0.000506859,-0.00963784],[0,-0.000553604,-0.00965043],[0,-0.000554325,-0.00963904],[0,-0.000499662,-0.00964459],[0,-0.000514868,-0.00962451],[0,-0.000561734,-0.00963722],[0,-0.000515827,-0.00965465],[0,-0.000519338,-0.00964659],[0,-0.000557512,-0.00963459],[0,-0.00051458,-0.00966089],[0,-0.000533118,-0.009643],[0,-0.000561333,-0.00965345],[0,-0.000584489,-0.00965961],[0,-0.000546021,-0.00964629],[0,-0.000555857,-0.00964245],[0,-0.000536812,-0.00961689],[0,-0.000570462,-0.00963899],[0,-0.000532813,-0.00963186],[0,-0.000563543,-0.00962475],[0,-0.000561559,-0.00962203],[0,-0.000525699,-0.00961292],[0,-0.000516837,-0.00962413],[0,-0.000523157,-0.00962436],[0,-0.00052429,-0.00964458],[0,-0.000514949,-0.00962932],[0,-0.000466287,-0.00961206],[0,-0.000544432,-0.00964529],[0,-0.000511904,-0.00963653],[0,-0.000516473,-0.00963743],[0,-0.000558109,-0.00962767],[0,-0.000517235,-0.00963725],[0,-0.000551676,-0.00963533],[0,-0.000579437,-0.00967226],[0,-0.000519596,-0.00962356],[0,-0.000547063,-0.009652],[0,-0.000561414,-0.00968008],[0,-0.000551386,-0.0096328],[0,-0.000556077,-0.00966545],[0,-0.000541396,-0.00962341],[0,-0.000575943,-0.00965069],[0,-0.00057 [... cut ...]
5455021 [ Flash,INF]: Erasing 64kB block at 655360
5455022 [ Flash,INF]: Erasing 64kB block at 720896
5455023 [ Flash,INF]: Erasing 64kB block at 786432
5455024 [ Flash,INF]: Erasing 64kB block at 851968
5455025 [ Flash,INF]: Erasing 64kB block at 917504
5455026 [ Flash,INF]: Erasing 64kB block at 983040
5455029 [   App,INF]: Writing firmware packet at address 0
5455031 [ Flash,ERR]: Verification error
```
I should probably try to power cycle the device but I'm not in the same building as the libreVNA so this will be something to try tomorrow.

jankae, 2022-06-26:
> I'm not sure what the first value in each tuple is supposed to represent. In the beginning of the sweep, it's zero, later on it's one, and in some sweeps it gets to two.

I would guess even in CW mode this field should still be equal to the frequency for the sweep? Perhaps an argument could be made to have the first column be time rather than frequency in CW mode.

It is the time, but without any digits after the decimal point. This is fixed in 36f826f7.

> At a random point in time I did also start getting this output in the device log

Almost certainly caused by an unfortunate mistake of mine in Delay::us (mixing different integer widths). This is also the cause for your failed firmware update. I just caught this myself minutes ago and am currently testing the fix. You might have to attempt the firmware update multiple times, but eventually it should work again.

Sorry for all that trouble, I guess I wasn't really paying close enough attention when implementing this feature.

> I also tried with the latest Windows GUI build on master but got an error on startup.

An issue with the build system (https://github.com/jankae/LibreVNA/commit/f9206c8b88091cf98f49ee79c6da5d9f861407ba#commitcomment-76940057), if you can compile it yourself, the error should disappear.

jankae, 2022-07-09:
Just wanted to let you know that the build error is fixed (latest windows on master works again) and I am also not aware of any issues with the firmware anymore. If you want, you can give this another try. Of course I can't promise you that there actually are no more bugs but at least I didn't encounter any in the last days while testing the zero span sweep.

Ivesvdf, 2022-07-09:
Great, thank you, I'll test tomorrow, I expect the ticket will be closed
then!

jankae, 2022-07-17:
Any update? If you haven't run into any new bugs, I'd like to close this

Ivesvdf, 2022-07-17:
I agree. Seems to work fine, thank you. Closing!

closed 2022-07-17 by Ivesvdf