jankae/LibreVNA #127: Marker display order

nbgsmk, 2022-07-08:
### LibreVNA Version

LibreVNA Version (64 bit): 1.4.0-ade970b35
OS: Windows 10 Version 2009
CPU Arch: x86_64

### Steps to reproduce

- right click on a graph -> add marker here
- repeat a few times, choosing a higher frequency along the way
- at some point, markers on the graph will become eg. 1, 2, 3, but on the right area their order is 2, 3, 1 (not consecutive)
- example: 
[image]

### Expected behavior

Marker data in the right area should be ordered by:
- creation sequence (most preferably).
- by increasing frequency (less preferable). A developer may wish to group several markers in a certain way, which is not dependent on increasing frequency. This is why I consider the first option is better.

### Extra information & Setup and Calibration files

n/a

### Confirming that you haven´t forgot to include configuration files (if applicable)

- [ ] Setup file
- [ ] Calibration file
- [ ] Touchstone file of measurements

jankae, 2022-07-09:
Thanks for the inspiration, I have added a sort order option in the preferences:
[image]

Previously, the markers were shown in the order they are stored in a std::set. Essentially, the order depended on the location of allocated memory for the markers, so pretty much random.

nbgsmk, 2022-07-09:
When dragging the marker on the graph in x-coordinate mode, the data display gets sorted on the fly. 
It's so cool, I love it!

closed 2022-07-09 by nbgsmk