jankae/LibreVNA #128: Marker temporary hide

nbgsmk, 2022-07-08:
### Summary

I tend to have a lot of markers in my graphs. Sometimes handling them all becomes tedious and the screen becomes cluttered. Especially when using bandpass, combined to other 'characteristic' points of interest nearby.
It would be a great help to temporary hide any marker 'triangle' from the graph (similarly as the traces can be hidden). The textual data on the right is ok to remain.
For marker groups such as bandpass, it would be sufficient to hide the whole group at once (hide each separately is also ok, whichever is easier to implement)

### Paths to implement it

Excuse my poor copy-pasting in paint, but something like this:

[image]

### Extra information

n/a

jankae, 2022-07-09:
Done, see 2954406
[image]

nbgsmk, 2022-07-09:
Great, I love it! Works like a charm. Closing.

closed 2022-07-09 by nbgsmk